evereux/pycatia #216: [BUG] CATPart, CATProduct documents.read issue

gggstamm, 2024-06-04:
**Describe the bug**
I'm facing this issue:
  File "...\pycatia\in_interfaces\documents.py", line 309, in read
    return document_types[doc_suffix](read_doc_com)
           ~~~~~~~~~~~~~~^^^^^^^^^^^^
KeyError: 'CATPart'

**To Reproduce**
Really simple script
    catia_com = catia()
    documents = catia_com.documents
    documents.read(str(c:\hello.CATPart))

It is working if I'm using 
documents.new_from
but it does no have exactly the same behavior.

I think that the issue here:
https://github.com/evereux/pycatia/blob/9f48c90e11539d3e6ae387ca3244a10e18ef8160/pycatia/in_interfaces/documents.py#L309
is that CATPart key is not present here
https://github.com/evereux/pycatia/blob/9f48c90e11539d3e6ae387ca3244a10e18ef8160/pycatia/types/document.py#L57
the key present is 'Part'

**Expected behavior**
no error !!

**Desktop (please complete the following information):**
 - OS: Windows 10
 - CATIA V5 version V5-R27
 - pycatia version : 0.7.0

evereux, 2024-06-04:
Yeah on the face of it this looks like a bug introduced when I re-wrote how document_types are handled recently. D'oh.

I'll get this fixed, probably at the weekend, and roll out a new version.

Thanks for reporting.

gggstamm, 2024-06-04:
> Yeah on the face of it this looks like a bug introduced when I re-wrote how document_types are handled recently. D'oh.
> 
> I'll get this fixed, probably at the weekend, and roll out a new version.
> 
> Thanks for reporting.

thanks for your very quick answer... and thanks in advanced for having a look
strange that nobody noticed the problem before
in the mean time, I went back to v0.6.8 ... that is working fine for me

evereux, 2024-06-06:
0.7.1 has now been released and should fix this issue.

Please note that the `Documents.read()` and `Documents.open()` now expect a pathlib.Path object. Strings should still work though.

evereux, 2024-06-14:
0.7.2 has now been released. This will actually now accept strings too.

closed 2024-06-14 by evereux

gggstamm, 2024-06-21:
Upgraded to 0.7.2... it is working fine.
many thanks