evereux/pycatia #217: [QUESTION]

ozserhatt, 2024-06-12:
Hello dear friends and evereux,
I can not get item that i selected element part or product.
I sent a code from win32com.client, I can not reach the referenceproduct in the SelectedElement Class.
How can I fix it. Thanks everything

```python
from win32com.client import Dispatch

# CATIA uygulamasını başlat
CATIA = Dispatch("CATIA.Application")

# Aktif belgeyi al
active_doc = CATIA.ActiveDocument

# Seçimi al
selection = active_doc.Selection

# Bilgilendirme mesajı
print("Please select a Part or Product!")

# Seçim için giriş nesneleri
input_objects = ["Product", "Part"]

# Seçimi yap
status = selection.SelectElement2(input_objects, "Please select main Product!", True)

# Seçim başarılıysa
if status:
    selected_doc = selection.Item(1).Type
    instance = selection.Item(1).Value

    if selected_doc == "Product":
        product_sel = selection.Item(1).Value
        selection.Clear()
        prod_ref = product_sel.ReferenceProduct.Parent
        if ".CATPart" in prod_ref.FullName:
            pass
           
        elif ".CATProduct" in prod_ref.FullName:
            pass
           
    else:
       
        pass
else:
   
    pass
```

ozserhatt, 2024-06-12:
I am sorry dear brother, I think it will be easy than explain my code. OK.
This is my code and I can not reach the reference_product 
Can you help me ?

```python
from pycatia import catia
from pycatia.in_interfaces.documents import Documents
from pycatia.product_structure_interfaces.product_document import ProductDocument
from pycatia.product_structure_interfaces.product import Product
from pycatia.mec_mod_interfaces.part_document import PartDocument
from pycatia.in_interfaces.selected_element import SelectedElement
from pycatia.in_interfaces.selection import Selection

caa = catia()
application = caa.application
doc = application.documents
active_document = application.active_document.com_object
selection = Selection(active_document.selection)
filter_ = ("Product", "Part")

status = selection.select_element2(filter_, 'Select your Products', False)
if status:
    get_element = Product(selection.item2(1))
    if get_element.reference_product.type == "CATPart":
        print("Selected Element is PART")
    if get_element.reference_product.type == "CATProduct":
        print("Selected Element is PRODUCT")
``` 

```python

return Product(self.product.ReferenceProduct)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'SelectedElement' object has no attribute 'ReferenceProduct'

```

evereux, 2024-06-12:
> Was issue #217 resolved for you?

Sorry, I meant: was issue #215 resolved for you?

I'll take a look at this shortly.

ozserhatt, 2024-06-12:
It is not about the #215.
I was not ask about the selection before already.
I can get item
 if active_document.product
but I can not get item
if active_document.selection

evereux, 2024-06-12:
I provided help for you yet you provided no feedback and didn't close the issue. I don't know if it was resolved.

ozserhatt, 2024-06-13:
Problem solved, thanks a lot brother.. by the way, I would like to meet with you if you want. I am a mechanical engineer and I hard working a project. Maybe I will tell you about that or other things. Thanks again.

closed 2024-06-14 by evereux