evereux/pycatia #218: [QUESTION] select_element3

ozserhatt, 2024-06-14:
Hello dear brother,
I can create multi section on the tree with select_element3 function.
If I choose one part, I can see on console " This is a CATPart" OK.  But ;
If I choose two parts I am getting the bug is :
```python
return SelectedElement(self.selection.Item2(i_index))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<COMObject <unknown>>", line 2, in Item2
pywintypes.com_error: (-2147352567, 'Özel durum oluştu.', (0, None, 'The iIndex value is not correct. Please give a value between 1 and 0', None, 0, -2147467259), None)
``` 

```python
from pycatia import catia
from pycatia.in_interfaces.selection import Selection
from pycatia.product_structure_interfaces.product import Product
from pycatia.mec_mod_interfaces.part_document import PartDocument
from pycatia.product_structure_interfaces.product_document import ProductDocument

caa = catia()
application = caa.application
doc = application.documents
active_document = application.active_document.com_object
selection = Selection(active_document.selection)

filter_ = ("Product", "Part")
status = selection.select_element3(filter_, 'Select your Products', False, 2, False)
if status:
    for selected_element in range(1, selection.count2 + 1):
        type = selection.item2(selected_element).type
        value = selection.item2(selected_element).value

        if type == 'Product':
            selection.clear()
            product = Product(value.com_object)
            if '.CATPart' in product.full_name:
                print('This is a CATPart.')

            if '.CATProduct' in product.full_name:
                print('This is a CATProduct.')

```

evereux, 2024-06-14:
There is a bug in the code for multiple selections and this is something you should be able to debug yourself. 😉  

There's a line in there I added during testing for #217 which I should have removed, or at least, placed earlier in the script.

You are iterating through the selected elements and getting items using `selection.item2(n)`. Look what happens to the `selection` object in the for loop and try and understand why the error is being thrown.

If you really can't figure it out I'll tell you what it is but try and figure this out on your own. 👍

ozserhatt, 2024-06-14:
:):):):) hahahah , maan, that was funny :) I remove the line "selection.clear()" Thank you so much

evereux, 2024-06-14:
Good stuff! Glad you figured it out by yourself.

closed 2024-06-25 by evereux