evereux/pycatia #220: [BUG]

RebelYoung, 2024-06-28:
**Describe the bug**
Fail to create a “Join” with 'hsf.add_new_join'

**To Reproduce**
1. creating 4 lines with 'hsf.add_new_line_pt_pt', 4 lines closure. A quad.
append line_list 'lines=[]'.
2. join=hsf.add_new_join(lines[0],lines[1])
join=hsf.add_new_join(j,lines[2])
join=hsf.add_new_join(j,lines[3])
part.update()
However, there is no “Join.1” element in CATIA.

**Expected behavior**
A clear and concise description of what you expected to happen.
Create a “Join.1” and revolve 360 degree into “Volume Revolve”.

Please help me, thankyou very much.
**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: [ Windows 10]
 - CATIA V5 version [e.g. V5.21]
 - pycatia version [e.g. pycatia 0.7.2]

**Additional context**
Add any other context about the problem here.

evereux, 2024-06-28:
Take a look at the [examples](https://github.com/evereux/pycatia/tree/master/examples)regarding the `HybridShapeFactory`.

What I think you might be missing is appending the `HybridShape` to the `HybridBody`.

If you're still having a problem please provide your **complete** script so that I may run it.

RebelYoung, 2024-07-01:
Thank you. It fixed when adding 'geom_set.append()'.

Now, I need the function ' Volume Revolve', but I can't find it in hsf class.

Do you know how to realize it ?
[image]

RebelYoung, 2024-07-01:
> Thank you. It fixed when adding 'geom_set.append()'.
> 
> Now, I need the function ' Volume Revolve', but I can't find it in hsf class.
> 
> Do you know how to realize it ?
> [image]
> 
> 

add_new_revol?

RebelYoung, 2024-07-01:
> > Thank you. It fixed when adding 'geom_set.append()'.
> > 
> > Now, I need the function ' Volume Revolve', but I can't find it in hsf class.
> > 
> > Do you know how to realize it ?
> > [image]
> > 
> > 
> 
> add_new_revol?

@evereux  Hello, I choose a way,
sf=part.shape_factory
revol=hsf.add_new_revol() ,
geom.append_hybrid_shape(revol),
sf.add_new_close_surface(Reference(revol.com_object))，
but error happend in 'sf.add_new_close_surface()'
[image]

evereux, 2024-07-01:
Please don't post screenshots unless neccessary.

Again, please post your **complete** script so I may run it. This is not paid support and getting a complete script, that isn't working, saves me time for a number of reasons.

When posting code please wrap it in three back ticks so make it more readable and so that we can get proper indenting. See the guide on writing [markdown](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax).

[HybridShapeFactory.add_new_revol()](https://github.com/evereux/pycatia/blob/5eb11e40795c0d2700351ebeba9d4e7d22acdffd/pycatia/hybrid_shape_interfaces/hybrid_shape_factory.py#L5389)
takes 4 arguments yet you haven't passed any to it?

**If you're struggling to figure out how to do something, record a macro. This will give you excellent clues on what you should be doing.**

The following will do what you're asking for.

```python

"""
Adds a HybridShapeFactory Revolution.

Requires a geometrical set called ConstructionGeometry and within that a line named Line.1 and sketch named Sketch.1

"""

from pycatia import catia
from pycatia.mec_mod_interfaces.part_document import PartDocument

caa = catia()
part_document:PartDocument = caa.active_document
part = part_document.part
hsf = part.hybrid_shape_factory
hbs = part.hybrid_bodies

construction_geometry = hbs.item("ConstructionGeometry")

construction_sketches = construction_geometry.hybrid_sketches
sketch_to_revolve = construction_sketches.item("Sketch.1")

hybrid_shapes_construction = construction_geometry.hybrid_shapes
line_axis = hybrid_shapes_construction.item("Line.1")

ref_sketch = part.create_reference_from_object(sketch_to_revolve)
ref_axis = part.create_reference_from_object(line_axis)

revolution = hsf.add_new_revol(ref_sketch, 180, 0, ref_axis)
revolution.context = 1

construction_geometry.append_hybrid_shape(revolution)

part.update()
```

RebelYoung, 2024-07-02:
> Please don't post screenshots unless neccessary.
>
> Again, please post your **complete** script so I may run it. This is not paid support and getting a complete script, that isn't working, saves me [...]

@evereux I am sorry, there is no script because the computer is offline.
Now, I coded on mobile again as follow:

```python
import pycatia
from pycatia.in_interfaces.reference import Reference

caa=pycatia.catia()
application=caa.application 
documents =caa.documents
part_document:PartDocument = documents.add("Part") 
part=part_document.part
shape_factory = part.shape_factory
hybrid_shape_factory = part.hybrid_shape_factory 
bodies = part.bodies 
part.update()

hsf=hybrid_shape_factory
hybrid_bodies = part.hybrid bodies geom_set = hybrid_bodies.add( )
geom_set.name = "Construction_Geometry"
pt_list=[[(1,0,0),(1.1,0,0)],[(1.1,0,0),(1.1,0,1)],[(1.1,0,1),(1,0,1)],[(1,0,1),(1,0,0)]]
1ines-[]	
for i in range(4):	
    co_ord_1 = pt_list[i][0]	
    co_ord 2 = pt_list[i][1]
    point_1 = hsf.add_new_point_coord(co_ord_1[8],co_ord_1[1],co_ord_1[2]) 
    point_1_reference = Reference(point_1.com_object)
    point_2 = hsf.add_new_point_coord(co_ord_2[0],co_ord_2[1], co_ord_2[2]) 
    point_2_reference = Reference(point_2.com_object)
    geom_set.append_hybrid_shape(point_1)
    geom_set.append_hybrid_shape(point_2)
    line = hsf.add_new_line_pt_pt(point_1_reference, point_2_reference)
    geom_set.append_hybrid_shape(line)
    lines .append(line)

j=hsf.add_new_join(lines[0],lines[1]) j=hsf.add_new_join(j,lines[2]) j=hsf.add_new_join(j,lines [3]; geom_set.append_hybrid_shape(j)
part.update( )

co_ord_1 = (0,0,0) 
co_ord_2 =(0,0,1)
point_1 = hsf.add_new_point_coord(co_ord_1[0], co_ord_1[1],co_ord_1[2]) 
point_1_reference = Reference(point_1.com_object)
point_2 = hsf.add_new_point_coord(co_ord_2[0], co_ord_2[1],co_ord_2[2]) 
point_2_reference = Reference(point_2.com_object)
geom_set.append_hybrid_shape(point_1)
geom_set.append_hybrid_shape(point_2)	

line_z = hsf.add_new_line_pt_pt(point_1_reference, point_2_reference)
geom_set.append_hybrid_shape(line_z) part.update( )

j_reference = Reference(j.com_object) z_ref=Reference(line_z.com_object)
j_revol=hsf.add_new_revol(j_reference,180,180,z_ref) geom_set.append_hybrid_shape(j_revol) part.update( )

# sf close surface error
j_revol_ref = part.create_reference_from_object(j_revol)
sf_close_surface = shape_factory.add_new_close_surface(j_revol_ref)

```

RebelYoung, 2024-07-02:
@evereux 
`j_revol_ref = part.create_reference_from_object(j_revol)`
or
`j_revol_ref1 = Reference(j_revol.com_object)`

Nether created close surface by   `shape_factory.add_new_close_surface()`.

evereux, 2024-07-02:
You coded that on mobile? I'm impressed. There were a couple of errors I had to fix but otherwise good work!

Anyways, there was a couple of steps missing related to making the `PartBody` the `in_work_object` prior to closing the surface.

Also, I changed the way you made the references in a number of places, instead using `part.create_reference_from_object()`. It wasn't wrong what you have done but this way is a bit nicer to use.

```python

import pycatia
from pycatia.in_interfaces.reference import Reference
from pycatia.mec_mod_interfaces.part_document import PartDocument

caa = pycatia.catia()
application = caa.application
documents = caa.documents
part_document:PartDocument = documents.add("Part")
part = part_document.part
shape_factory = part.shape_factory
hybrid_shape_factory = part.hybrid_shape_factory
bodies = part.bodies
part_body = bodies.item('PartBody')

hsf = hybrid_shape_factory
hybrid_bodies = part.hybrid_bodies
geom_set = hybrid_bodies.add( )
geom_set.name = "Construction_Geometry"
pt_list=[
    [(1,0,0),(1.1,0,0)],
    [(1.1,0,0),(1.1,0,1)],
    [(1.1,0,1),(1,0,1)],
    [(1,0,1),(1,0,0)]
]
lines=[]
for i in range(4):
    co_ord_1 = pt_list[i][0]
    co_ord_2 = pt_list[i][1]

    point_1 = hsf.add_new_point_coord(co_ord_1[0],co_ord_1[1],co_ord_1[2])
    point_1_reference = part.create_reference_from_object(point_1)

    point_2 = hsf.add_new_point_coord(co_ord_2[0],co_ord_2[1], co_ord_2[2])
    point_2_reference = part.create_reference_from_object(point_2)

    geom_set.append_hybrid_shape(point_1)
    geom_set.append_hybrid_shape(point_2)

    line = hsf.add_new_line_pt_pt(point_1_reference, point_2_reference)
    geom_set.append_hybrid_shape(line)
    line_reference = part.create_reference_from_object(line)
    lines.append(line_reference)

j = hsf.add_new_join(lines[0],lines[1])
j = hsf.add_new_join(j,lines[2])
j = hsf.add_new_join(j,lines [3])
geom_set.append_hybrid_shape(j)

part.update( )

co_ord_1 = (0,0,0)
co_ord_2 = (0,0,1)

point_1 = hsf.add_new_point_coord(co_ord_1[0], co_ord_1[1],co_ord_1[2])
point_1_reference = part.create_reference_from_object(point_1)

point_2 = hsf.add_new_point_coord(co_ord_2[0], co_ord_2[1], co_ord_2[2])
point_2_reference = part.create_reference_from_object(point_2)

geom_set.append_hybrid_shape(point_1)
geom_set.append_hybrid_shape(point_2)

line_z = hsf.add_new_line_pt_pt(point_1_reference, point_2_reference)
geom_set.append_hybrid_shape(line_z)
part.update( )

j_reference = part.create_reference_from_object(j)
z_ref = part.create_reference_from_object(line_z)
j_revol = hsf.add_new_revol(j_reference,180,180,z_ref)
geom_set.append_hybrid_shape(j_revol)
part.update( )

# sf close surface error
j_revol_ref = part.create_reference_from_object(j_revol)
part.in_work_object = part_body
sf_close_surface = shape_factory.add_new_close_surface(j_revol_ref)

part.update()
```

evereux, 2024-07-02:
Also, just to reiterate again, I figured this out by creating a macro doing that task I was trying to figure out.

Reading the results of the macro helped point me in the right direction.

The macro creation tool baked into CATIA V5 is very helpful!

RebelYoung, 2024-07-02:
> You coded that on mobile? I'm impressed. There were a couple of errors I had to fix but otherwise good work!
>
> Anyways, there was a couple of steps missing related to making the `PartBody` the [...]

@evereux 👍👍👍👏👏👏Thank you very much for your help. even if I do not understand why should `part.in_work_object = part_body`? When I operated manually not changing work object，the close surface also could be created in geometry set.

evereux, 2024-07-02:
It's just one of those quirks of the CATIA V5 interface.

When you added the new `GeometricalSet` and added items to it that  `GeometricalSet` that automatically became the `in_work_object`.

You can't add a `ShapeFactory` close surface to a `GeometricalSet`, it **has** to be a `Body`. 

There is also a `HybridShapeFactory` close surface which you could add to a `GeometricalSet`.

RebelYoung, 2024-07-02:
> It's just one of those quirks of the CATIA V5 interface.
>
> When you added the new `GeometricalSet` and added items to it that  `GeometricalSet` that automatically became the `in_work_object`.
>
> You [...]

 
😂I haven't found the function like `add_new_close_surface()` in `hybrid_shape_factory.py` class.

evereux, 2024-07-02:
You're right. It's `add_new_volume_close_surface` I'm thinking of that. It's this that will allow you to add it to a `GeometricalSet`.

Weird logic but not a pycatia thing.

closed 2024-07-04 by evereux