evereux/pycatia #221: auto set "DisplayFileAlerts=False" when use "export_data (... overwrite=True)"

Zcaic, 2024-06-29:
Hello, I found it's good that when I use `save_as(...  overwrite=True)` ,the `DisplayFileAlerts` will be auto set to be `False`.
But when I export `stp` format with command `export_data (... overwrite=True)`, the `DisplayFileAlerts` won't be set to be `False`.

It is easy to change below codes in `document.py` file
```
   line 476     if overwrite is False and file_name.is_file():
   line 477         raise FileExistsError(f'File: {file_name} already exists. '
   line 478                               f'Set overwrite=True if you want to overwrite.')
```
to:
```
        if overwrite is False:
            if path_file_name.is_file():
                raise FileExistsError(f'File: {path_file_name} already exists. '
                                      f'Set overwrite=True if you want to overwrite.')
        else:
            self.document.Application.DisplayFileAlerts = False
```

evereux, 2024-07-01:
I've submitted a fix for this which will be in 0.7.3 release.

Thanks for reporting.

evereux, 2024-07-13:
0.7.3 has now been released.

closed 2024-07-13 by evereux