evereux/pycatia #222: [QUESTION] IMPORTANT QUESTION

ozserhatt, 2024-07-01:
Dear Friends !!!

CAN I EXPORT THE MACRO ON CATIA AS .VBS ?

evereux, 2024-07-01:
You can run scripts using [caa.application.system_service.evaluate()](https://github.com/evereux/pycatia/blob/5eb11e40795c0d2700351ebeba9d4e7d22acdffd/pycatia/system_interfaces/system_service.py#L63) or [caa.application.system_service.execute_script()](https://github.com/evereux/pycatia/blob/5eb11e40795c0d2700351ebeba9d4e7d22acdffd/pycatia/system_interfaces/system_service.py#L202).

`caa.application.system_service.evaluate()` will need you to read the script and pass it as string.

`caa.application.system_service.execuate_script()` is probably what you want.

ozserhatt, 2024-07-02:
I have a .vbs document in my desktop. It is a macro that design a cube on Catia. I can choose my file with :
```python
file_path = filedialog.askopenfilename(filetypes=[('VBS Files (*.vbs)|*.vbs', "")])
``` 

But how can i run on Catia  ?

ozserhatt, 2024-07-02:
param str i_library_name:
param int i_type: enum cat_script_library_type
param str i_program_name:
param str i_function_name:
**param tuple i_parameters:**

what does it mean parameters ??  I have a .vbs document that macro. I only run it

evereux, 2024-07-02:
They would be parameters you want to pass into your script. If you don't need to pass any try passing an empty list `[]`.

There are several places in the pycatia application where `system_service.evaluate()` is used. Take a look at those usages for reference to do a similar task but passing the script as a string.

ozserhatt, 2024-07-02:
Yeah! I tried that like what you said. Already i find right. But i have another problem.

```python
application.system_service.execute_script(r"C:\Users\serhat.oz\Desktop\library-vbs", 1, "Macro3", "CATMain", params)
``` 

Error: (-2147352567, 'Özel durum oluştu.', (0, None, 'ExecuteScript(C:\\Users\\serhat.oz\\Desktop\\library-vbs, Macro3, CATMain)\nUnknown macro program "Macro3"\n', None, 0, -2147467259), None)

evereux, 2024-07-02:
I've never used it and don't have a VBS I can test it with.

Are you sure your script name is correct? It has no filename extension.

ozserhatt, 2024-07-02:
There is bug that i have : 
```python
Error: (-2147352567, 'Özel durum oluştu.', (0, None, 'ExecuteScript(Part6.CATPart, Macro2.catvbs, CATMain)\nUnknown macro library "Part6.CATPart"\n', None, 0, -2147467259), None)

``` 

[image]

evereux, 2024-07-02:
From the automation documentation:

```
                | o Func ExecuteScript(CATBSTR iLibraryName,
                | CatScriptLibraryType iType,
                | CATBSTR iProgramName,
                | CATBSTR iFunctionName,
                | CATSafeArrayVariant iParameters) As CATVariant
                | 
                |     Executes a scripted function.
                | 
                |     Parameters:
                | 
                |         iLibraryName
                |             The library in which the script is contained 
                |         iLibraryType
                |             The type of the library 
                |         iProgramName
                |             The name of the program in the library 
                |         iFunctionName
                |             The name of the function to invoke 
                |         iParameters
                |             An array of parameters for the function 
                |         oResult
                |             The value returned by the function (if any) 
                | 
                |     Example:
                |         This example executes the function CATMain in the program Macro1.catvbs
                |         contained by Part1.CATPart
                | 
                |          Dim params()
                |          CATIA.SystemService.ExecuteScript"Part1.CATPart",
                |          catScriptLibraryTypeDocument, "Macro1.catvbs", "CATMain",
                |          params
```

You would only reference the `CATPart` if the script is contained within the `CATPart` itself.

 I would do some googling on how other people have done this using CATScript / VB or whatever.

Hopefully someone else can help you. I don't have anymore time to spend on this.

ozserhatt, 2024-07-02:
It's OK!!! THAT IS RUNNING NOW ! I DO NOT KNOW HOW IT IS BUT THIS CODE IS RUNNING MACRO !

```python
import os
import sys
from tkinter import Tk, filedialog
from pycatia import catia
from pycatia.mec_mod_interfaces.part_document import PartDocument
from pycatia.product_structure_interfaces.product_document import ProductDocument
from pycatia.system_interfaces.system_service import SystemService

def run_catia_macro():
    try:
        params = []

        application.system_service.execute_script("Part1.CATPart", 0, "Macro1.catvbs", "CATMain", params)

    except Exception as e:
        print(f"Error: {str(e)}")
    finally:
        pass

caa = catia()
application = caa.application
doc = application.documents
active_document = ProductDocument(application.active_document.com_object)
product = active_document.product
if product.is_catpart():
    cat_type = product.reference_product.type == "CATPart"
    print(product.reference_product.name)
    part_document = PartDocument(doc.item(product.reference_product.file_name).com_object)
    part = part_document.part
    run_catia_macro()
```

evereux, 2024-07-02:
That's interesting. 

Glad you got it figured out and thanks for posting the working solution.

closed 2024-07-04 by evereux