evereux/pycatia #225: [BUG] can't open files

amirhosseinshakoori, 2024-07-07:
**Describe the bug**
Sometimes in version 0.7.2 the error "The document cannot be read" is displayed This error is only in this version and is not seen in other versions Even in some times when the program is run, this error is seen and sometimes there is no error

.

evereux, 2024-07-07:
That's generally an error message reported by CATIA V5 when the documents saved version is later than the version of CATIA V5 you are using.

I've just tried opening and closing a document a 1000 times and couldn't trigger the error.

If you don't think this is the issue please provide more information.

evereux, 2024-07-13:
Closing as no information provided on how to reproduce the issue.

closed 2024-07-13 by evereux