evereux/pycatia #226: [QUESTION] Is it possible to find MML links beetween two CATDrawing file?

Djey51, 2024-07-16:
Hi,
Everything is in the title.
I have one drawing. In this drawing, a view contening a component comming from an other catdrawing. So I have a MML between the two drawings.
Is there a way to find the name, at least of the l i nked drawing please?

Thanks a lot for your help

evereux, 2024-07-16:
Not sure. 

Have you tried googling to see how this would be done in VBA / CATScript?

evereux, 2024-07-16:
I think I did the same a few years back and couldn't find anything. Maybe someone else will be able to shed light on this?

I'll see if I can figure it out in a couple of days.

evereux, 2024-07-18:
Well, I at least learned about `DrawingViewGenerativeLinks` which works for views generated from a CATPart.

But as I'm sure you're already aware this doesn't work for a `View` linked to another `CATDrawing`. 

If I try to manually ` Edit > Links > Replace` on a  CATIA V5 itself throws up a warning message `Replace not allowed in that context for document ..`. Whatever `context` means in this context. 😝

Djey51, 2024-07-18:
Yes effectively, I was already aware of that and I think that DrawingViewGenerativeBehavior is more powerfull but limited to CATPart and CATproduct file too.

I'm pretty sure that catvba API dob't allow to perform that and we should switch to CAALanguage to be able to identify this links.

But I still hope that a smart brain in the community will find a trick way to reach the result

siddhu2310, 2024-08-20:
@Djey51 could you share more details, I will try to figure a possible way.
Do share some example files as well.

Djey51, 2024-08-20:
Hello @siddhu2310 ,
For exemple : I have got 3 files in the following zip file : 
- PART_L.CATPart --> a very simple Model
- EXEMPLE.CATDrawing --> a drawing with a component
- DRAWING.CATDrawing --> a drawing which is using the previous 2 files

When I work in DRAWING.CATDrawing if I edit the links I obtain the following picture 
[image]
We can see 2 kind of links : 
- MML link
- Viewlink link
If I open only the file DRAWING.CATDrawing, with pycatia, I'm able to find the information about the link Viewlink (at list the name of the part and if the link is not broken I can also obtain the full path of the PART_L.CATPart
But my issue is to find information on the MML link : for me it is not possible but if you find the solution you are definitively the best !!

Djey51, 2024-08-20:
I don't know why but I can't attach my zip file...

siddhu2310, 2024-08-22:
I can be able to get the below  names.
EXEMPLE
PART1

[image]

@Djey51  Could you tell me get the full path PART.

siddhu2310, 2024-08-22:
@Djey51 
Finally I got the solution:

Here is the code :

```python
from pycatia import catia
from pycatia.cat_smarteam_integ_interfaces.sti_engine import StiEngine,StiDBItem    
from pycatia.drafting_interfaces.drawing_document import DrawingDocument

caa = catia()
DrwDocument = DrawingDocument(caa.active_document.com_object)
sti_Engine = caa.get_item("CAIEngine")
sti_Engine = StiEngine(sti_Engine.com_object)
sti_DbItem  = sti_Engine.get_sti_db_item_from_any_object(DrwDocument)
sti_DbItem = StiDBItem(sti_DbItem.com_object)

sti_children = sti_DbItem.get_children()
print(sti_DbItem.get_children().count)

for i in range(1,sti_children.count+1):
    linked_item = sti_children.item(i).is_root()
    if linked_item:
        print(sti_children.item(i).get_document().name)
        print(sti_children.item(i).get_document_full_path())
```

Djey51, 2024-08-22:
Hi @siddhu2310,

Thanks for the code.
Unfortunally, I tested it but I obtain the following issue
[image]

Python 3.11
Pycatia 0.7.3
CatiaV5 6R2015SP4HF38
Windows10

Djey51, 2024-08-22:
Ok @siddhu2310 ,

 Yoyr code works very well if there are only links to CATPart.
But I obtain the issue as soon as the drawing have an other drawing linked thanks to a mml link.

In your previous answer you said you were able to find the two names EXEMPLE and PART1.

Could you provide me a code for that please?

You found a very simple way to list the part linked (better than my existing code) thanks for that.

BR

siddhu2310, 2024-08-22:
[image]

It should work, This is the output of the example file you shared.

Djey51, 2024-08-22:
It works!!
I don't know why but I need to modify a little bit to make it working on ly environment.
`
if sti_children_count > 0 :
    for i in range(1,sti_children_count+1):
        linked_item = sti_children.item(i).is_root()
        if linked_item:
            print(sti_children.item(i).get_document().name)
            print(sti_children.item(i).get_document_full_path())`

evereux, 2024-08-24:
Thanks for helping out here @siddhu2310. Interesting bit of code.

siddhu2310, 2024-08-28:
@Djey51 Close this Question

evereux, 2024-08-28:
> @Djey51 Close this Question

@Djey51 Please don't. 🤣  The code here uses a set of features of pycatia not covered before and I'd like to add this as an example / user script. This being open will remind me to do that. 😄

Djey51, 2024-08-28:
Hi @evereux,

If I have nothing to do, it is fine fot me🤭

Djey51, 2024-09-04:
I @siddhu2310 ,

As this ticket is not yet closed, I have an other question linked to this topic. 
We are able thanks to your code to obtain informations on Part and Drawing when links are still active : but how to display the same information as the interactive Catia window "Edit/Links" when the links are broken? How to get the link type and the name of the document targeted?
It should be a very complete and interesting example for the Pycatia show cases.
Best regards

Djey51, 2024-10-12:
Hello @siddhu2310 ,
An idea ?

siddhu2310, 2024-10-25:
Hi @Djey51 

I don't have time to check now, Will update you once I figured out.