evereux/pycatia #227: Access ParameterSets inside an HybridBody

chrisoraruf, 2024-07-18:
Hi guys,

the documentation looks really clear about the ParameterSets. Nevertheless, it looks like from the Part.parameters you can just go back to the RootParameterSet and then move forward inside of it. The thing is that I have this kind of structure.

Part
- Parameters
     - ParameterSet
     - ParameterSet
- Relations
- HybridBody
     - HybridBody
     - ParameterSet
     - HybridBody

And I have several of this ParameterSet inside HybridBodies. From Part.parameters I can go back to Parameters and then move inside its ParameterSets, but no way I can find a solution to see the other ParameterSets. 

You guys have hints?

Bests!

evereux, 2024-07-20:
You could use [Parameters.sub_list](https://github.com/evereux/pycatia/blob/77ef90b72f204922fa8f53fa47978a636113723a/pycatia/knowledge_interfaces/parameters.py#L521) by passing the `HybridBody` object to it.

But, this will give you a list of all parameters in the `HybridBody` such as parameters used to define geometry in the `HybridBody`. Easily filtered if you know the `ParameterSet` name.

Discovered this myself by googling how this would be done in VB. I couldn't find any more direct approach unfortunately. If there is one it's not obvious to me.

evereux, 2024-09-16:
@siddhu2310 

Do you really need examples files? @chrisoraruf has provided enough information in their initial post.

closed 2025-08-29 by evereux