evereux/pycatia #229: [BUG] Drawing Dimensions.Add2 Error

siddhu2310, 2024-08-09:
**Describe the bug**
I have created two points in the Drawing, then tried to generate the Dimensions for it, Just I need to show the Distance Between the Points.

**Error:**
'''
Traceback (most recent call last):
  File "d:\Testing\Drwing\testdim.py", line 32, in <module>
    my_dimension.add2(dim_type, my_elements, selection_points, Line2, 0)
  File "C:\Python311\Lib\site-packages\pycatia\drafting_interfaces\drawing_dimensions.py", line 196, in add2
    self.drawing_dimensions.Add2(
  File "<COMObject <unknown>>", line 3, in Add2
TypeError: must be real number, not Point2D
'''

**Code**

```
from pycatia import catia
from pycatia.drafting_interfaces.drawing_dimensions import DrawingDimensions
from pycatia.drafting_interfaces.drawing_document import DrawingDocument
from pycatia.drafting_interfaces.drawing_view import DrawingView
from pycatia.enumeration.enumeration_types import cat_dim_type

caa = catia()
drawing_document: DrawingDocument = caa.active_document
drawing_sheet = drawing_document.sheets.active_sheet
active_view = drawing_sheet.views.active_view
active_view = DrawingView(active_view.com_object)
factory_2d = active_view.factory_2d

point_1 = factory_2d.create_point(40, 230)
point_2 = factory_2d.create_point(80, 210)

Line2 = factory_2d.create_line(50, 10, 50, 100)

dim_type = cat_dim_type.index("catDimDistance")

my_elements = (point_1, point_1)
selection_points = (0, 0, 0, 0) 

my_dimension = DrawingDimensions(active_view.dimensions.com_object)
my_dimension.add2(dim_type, my_elements, selection_points, Line2, 0)

drawing_sheet.force_update()
```

**Desktop (please complete the following information):**
 - OS: Windows 11
 - CATIA V5 version : V5-6R2020
 - pycatia version  : 0.7.2

siddhu2310, 2024-08-13:
**Equivalent CATIA VBA Code**

```
Sub CATMain()

    Dim Fact2D As Factory2D

    Dim documents1 As Documents
    Set documents1 = CATIA.Documents

    Dim drawingDocument1 As drawingDocument
    Set drawingDocument1 = documents1.Add("Drawing")

    drawingDocument1.Standard = catISO

    Dim drawingSheets1 As drawingSheets
    Set drawingSheets1 = drawingDocument1.Sheets

    Dim drawingSheet1 As drawingSheet
    Set drawingSheet1 = drawingSheets1.Item("Sheet.1")

    drawingSheet1.PaperSize = catPaperA3
    drawingSheet1.Orientation = catPaperLandscape

    Dim drawingViews1 As drawingViews
    Set drawingViews1 = drawingSheet1.Views
    
    'Sidd
    'Dim drawingView1 As drawingView
    Set drawingView1 = drawingViews1.Add("AutomaticNaming")

   
    
    
    Set Fact2D = drawingView1.Factory2D

    drawingView1.X = 420 / 2
    drawingView1.Y = 297 / 2

    drawingView1.FrameVisualization = False
    drawingView1.Scale2 = 3
    
    Dim drawingViewGenerativeLinks1 As DrawingViewGenerativeLinks
    Set drawingViewGenerativeLinks1 = drawingView1.GenerativeLinks

    Dim drawingViewGenerativeBehavior1 As DrawingViewGenerativeBehavior
    Set drawingViewGenerativeBehavior1 = drawingView1.GenerativeBehavior

    Dim partDocument1 As partDocument
    Set partDocument1 = documents1.Item("DEMO_1.CATPart")

    Dim part1 As part
    Set part1 = partDocument1.part

    Dim axisSystems1 As axisSystems
    Set axisSystems1 = part1.axisSystems

    Dim axisSystem1 As axisSystem

    Set axisSystem1 = axisSystems1.Item("Axis Sys06")
    

    drawingViewGenerativeBehavior1.SetAxisSysteme Nothing, axisSystem1

    Dim bodies1 As Bodies
    Set bodies1 = part1.Bodies

    Dim body1 As Body
    Set body1 = bodies1.Item("Part1")

    drawingViewGenerativeLinks1.AddLink body1
    
    
    
    Dim body2 As Body
    Set body2 = bodies1.Item("Part2")

    drawingViewGenerativeLinks1.AddLink body2

    drawingViewGenerativeBehavior1.DefineFrontView -0.087122, 0.996198, 0#, -0.002424, -0.000212, -0.999997
    drawingViewGenerativeBehavior1.Update

    drawingView1.Activate

    Dim P1X As Double, P1Y As Double
    Dim P2X As Double, P2Y As Double

    P1X = 15.00341
    P1Y = 10.40991
    P2X = 5.671
    P2Y = 14.00241

Dim Point1 As Point2D
Dim Point2 As Point2D
Dim Line2D As Line2D
Dim myElements
Dim selpoints
Dim MyDimension As DrawingDimension

 Set origine = Fact2D.CreatePoint(0, 0)
 Set PointP2 = Fact2D.CreatePoint(P2X, P2Y)
 Set PointP1 = Fact2D.CreatePoint(P1X, P1Y)
 
 Set Line2D = Fact2D.CreateLine(P2X, P2Y, P1X, P1Y)
 
 myElements = Array(origine, Line2D)
 selpoints = Array(0, 0, 0, 0)
 
 Set MyDimension = drawingView1.Dimensions.Add(catDimDistance, myElements, selpoints, catDimDistanceMin)

    
    drawingViewGenerativeBehavior1.ForceUpdate

    Dim specsAndGeomWindow1 As SpecsAndGeomWindow
    Set specsAndGeomWindow1 = CATIA.ActiveWindow

    Dim specsViewer1 As SpecsViewer
    Set specsViewer1 = specsAndGeomWindow1.ActiveViewer

    specsViewer1.Reframe
    
    

End Sub
```

siddhu2310, 2024-08-14:
I have Solved the Issue Myself, Just Passed Com _Object 

```
elements = (origine.com_object,line_2d.com_object)
```

evereux, 2024-08-14:
I'm re-opening this issue as you shouldn't have to do that as an end user. That should be done within the class method itself so I'll get this patched soon. However, the update will break your current solution.

evereux, 2024-08-14:
There's been some ninja editing here ... The original question was regarding the `DrawingDimensions.add2()` class method. Both require fixing.

evereux, 2024-08-14:
I've submitted fixes for both `DrawingDimensions.add()` and  `DrawingDimensions.add2()` to the developement branch which will be included in the next 0.74 release.

Here are complete working examples for both. These require an already open CATDrawing with an active view.

## DrawingDimensions.add()

```python
from pycatia import catia
from pycatia.enumeration.enumeration_types import cat_dim_type
from pycatia.enumeration.enumeration_types import cat_dim_line_rep
from pycatia.drafting_interfaces.drawing_document import DrawingDocument

caa = catia()
drawing_document: DrawingDocument = caa.active_document
sheets = drawing_document.sheets
sheet = sheets.active_sheet
views = sheet.views
active_view = views.active_view
factory_2d = active_view.factory_2d

line_1 = factory_2d.create_line(50, 10, 150, 10)
line_2 = factory_2d.create_line(50, 10, 120, 100)
elipse = factory_2d.create_ellipse(-40, 100, 120, 180, 120, 90, 0, 3)
point_1 = factory_2d.create_point(-10, 190)
point_2 = factory_2d.create_point(-120, 190)

catDimAngle = cat_dim_type.index('catDimAngle')
catDimAuto = cat_dim_line_rep.index('catDimAuto')

line_elements = (line_1, line_2)
selection_points_1 = (150, 10, 120, 100)

active_view.dimensions.add(catDimAngle, line_elements, selection_points_1, catDimAuto)

catDimLengthCurvilinear = cat_dim_type.index('catDimLengthCurvilinear')
catDimOffset = cat_dim_line_rep.index('catDimOffset')

var_elements = (point_1, point_2, elipse)
selection_points_2 = (0, 0, 0, 0)
active_view.dimensions.add(catDimLengthCurvilinear, var_elements, selection_points_2, catDimOffset)
```

## DrawingDimensions.add2()

```python

from pycatia import catia
from pycatia.enumeration.enumeration_types import cat_dim_type
from pycatia.enumeration.enumeration_types import cat_dim_line_rep
from pycatia.drafting_interfaces.drawing_document import DrawingDocument

caa = catia()
drawing_document: DrawingDocument = caa.active_document
sheets = drawing_document.sheets
sheet = sheets.active_sheet
views = sheet.views
active_view = views.active_view
factory_2d = active_view.factory_2d

point_1 = factory_2d.create_point(40, 230)
point_2 = factory_2d.create_point(80, 210)
line_1 = factory_2d.create_line(50, 10, 150, 10)

catDimDistance = cat_dim_type.index('catDimDistance')

point_elements = (point_1, point_2)
selection_points = (0, 0, 0, 0)

dimension = active_view.dimensions.add2(catDimDistance, point_elements, selection_points, line_1, 0)
```

closed 2024-08-14 by evereux

siddhu2310, 2024-08-19:
Thank you so much for the support @evereux 
When I can expect the 0.7.4 Release ?

evereux, 2024-08-19:
This has now been released.