evereux/pycatia #230: [BUG] Missing  scale_on_extremities in DrawingArrow

siddhu2310, 2024-08-19:
**Describe the bug**
scale_on_extremities property  is missing in the **DrawingArrow** Object

**Expected behavior**
These below property need to be added in the **drawing_arrow.py**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - CATIA V5-6R2020
 - pycatia version : pycatia 0.7.3

evereux, 2024-08-19:
According to the `V5Automation.chm` from `V5-6R2021` this was introduced in `V5-6R2018 GA`. However, I do have the `V5Automation.chm` from `V5-6R2018 GA` which was in fact used to generate pycatia. Must not have been properly documented at this stage by Dassault.

Anyways, this will get added but I'll need to go through all the new features and try and capture them too.  I'll also need to get pycatia to do a version check before the user runs the these new methods. Unless, I leave them to fail without telling the user why, maybe just comment on the fact in the doc string for the method? 

Unfortunately I don't currently have a way to test them anyway.

I'll need to think on this ....

siddhu2310, 2024-08-20:
[image]

### Property ScaleOnExtremities() As boolean
Returns or sets the scale on extremities mode.
**Example:**
This example retrieves the target element of head side on MyArrow drawing arrow.
```
 oScaleOnExtremities = MyArrow.ScaleOnExtremities
```

evereux, 2024-08-24:
I've now added a bunch of new Classes, methods, properties and enums introduced between R2018 and R2021 which are listed in the [CHANGELOG.md](https://github.com/evereux/pycatia/blob/developement/CHANGELOG.md) for the developement branch. 

One of these additions is `DrawingArrow.scale_on_extremities`.

This will be included in next release which will be 0.8.0.

Thanks for letting me know about this.

I should be able to get the `V5Automation.chm` for `CATIA V5-6R2023` in a couple of weeks so I'll add anything missing between 
`CATIA V5-6R2021` > `CATIA V5-6R2023` then.

siddhu2310, 2024-08-26:
@evereux

If Suppose user uses the lower version, notify them with error msg : These Methods or Class, Property not available with this CATIA version.

evereux, 2024-08-26:
What I ended up doing was simply log a message to the console if the CATIA version is lower than the requirement.

0.8.0 is now released.

closed 2024-08-28 by siddhu2310