evereux/pycatia #231: [FEATURE REQUEST] Modify catia settings

Djey51, 2024-09-16:
Hi guis,

I would like to manage the settings of the attached picture, thanks to a script. D o you have an idea on how to do that?
[image]

For information, I tried to export catscript file of this tab option but I obtain an empty sub.

Thanks

evereux, 2024-09-16:
I couldn't find anything myself.

I'll tag a few other users to see if they have any idea: @siddhu2310 @deloarts @Mithro86.

Mithro86, 2024-09-16:
At the moment I don't have access to CATIA so I can't check. 
But I know many of the settings are not accessible.
I did a (not so pretty) run on activating shareable licenses some years ago, using pywinauto.

Found here: https://github.com/Mithro86/CATIA-V5-scripts/blob/main/dynamic_license.py

Maybe it can be of help.

siddhu2310, 2024-09-17:
> Hi guis,
> 
> I would like to manage the settings of the attached picture, thanks to a script. D o you have an idea on how to do that? ![MultiLayerSetting](https://private-user-images.githubusercontent.com/61665076/367747963-5febc581-b264-497a-90bd-3b594b7422c6.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY1NTA0ODYsIm5iZiI6MTcyNjU1MDE4NiwicGF0aCI6Ii82MTY2NTA3Ni8zNjc3NDc5NjMtNWZlYmM1ODEtYjI2NC00OTdhLTkwYmQtM2I1OTRiNzQyMmM2LlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE3VDA1MTYyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY4NWViYTI3Y2YyNzVkYzg3Y2NlNWU3MzRjYjI0NmZlMjA1ZjVmZTFkMTcwMzIxNWNmMTQ2NTNlYjBhMjg0MzMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.fS6l_0tTs_KBoERvOSfhBzfoTHGLl8pMvwqYdBLssAo)
> 
> For information, I tried to export catscript file of this tab option but I obtain an empty sub.
> 
> Thanks

I have checked in the CATIA Documentation, It is not available .

Djey51, 2024-09-18:
Hi @siddhu2310 ,
I also checked the Catia documentation but for me,  it is not clear what is available or not around the settings.
If you have the specific chapter saying that it is not available, it interests me so much.
Thanks