evereux/pycatia #232: [QUESTION] User ref Properties

ptm-tm, 2024-10-03:
This is is not valid
```
product = document.product
user_properties = product.user_ref_properties
for i in range(1, user_properties.count + 1):
    print(user_properties.item(i).name,"=",user_properties.value)
```
but if we insert additional assignments - this code will be valid.
```
product = document.product
user_properties = product.user_ref_properties
for i in range(1, user_properties.count + 1):
    a=user_properties.item(i)
    print(a.name,"=",a.value)
```
its normally?

ptm-tm, 2024-10-03:
sry my mistake... need add item(i) to value

closed 2024-10-03 by ptm-tm

evereux, 2024-10-03:
You could do it like this:

```python

from pycatia import catia
from pycatia.product_structure_interfaces.product_document import ProductDocument

caa = catia()
product_document: ProductDocument = caa.active_document
product = product_document.product
user_ref_properties = product.user_ref_properties

for i in range(len(user_ref_properties)):
    index = i + 1
    print(f'{user_ref_properties.item(index).name}={user_ref_properties.item(index).value}')

```

Maybe a little easier to read? I should update the examples to reflect this when using range on a collection.

ptm-tm, 2024-10-04:
ye) its more pythonic but im create simple for quick test..

ptm-tm, 2024-10-04:
> You could do it like this:
>
> ```python
> from pycatia import catia
> from pycatia.product_structure_interfaces.product_document import ProductDocument
>
> caa = catia()
> product_document: ProductDocument = [...]

```
for i in user_ref_properties:
    print(i.name,i.value_as_string())
```
its more readable... but i cant use 
`i.value
`