evereux/pycatia #234: self.part.update Error

FaiqShahbaz, 2024-11-18:
I am encountering part update error on most of the designs i am doing via pyCatia, when run on different pc its working alright (which means the code is okay). i have attached python code and the error i am encountering.

Code:
# ----- Packages
# --------------------------------------------------------------------------- #
from pycatia import catia
from pycatia.mec_mod_interfaces.part_document import PartDocument

#
# =========================================================================== #
# ----- CATIA variables
# =========================================================================== #
#
caa = catia()
application = caa.application
documents = application.documents
if documents.count > 0:
    for document in documents:
        document.close()
        
# Create Part
documents.add('Part')

# Get references to document, parts and tools to create geometries
document: PartDocument = application.active_document        # Get active document
part = document.part                                        # Get Part
partbody = part.bodies[0]                                   # Get default PartBody by index
# partbody = part.bodies.item("PartBody")                     # Get default PartBody by name
sketches = partbody.sketches                                # Get sketches on PartBody 
hybrid_bodies = part.hybrid_bodies                          # Get Hybrid Bodies 
hsf = part.hybrid_shape_factory                             # Get Hybrid Shape Factory 
shpfac = part.shape_factory                                 # Get Shape Factory 
selection = document.selection                              # Get document selection

# Get references to main planes
plane_XY = part.origin_elements.plane_xy
plane_YZ = part.origin_elements.plane_yz
plane_ZX = part.origin_elements.plane_zx

# Set PartBody as a working object
part.in_work_object = partbody

# Arbitrary dimension
L = 100

# Create a new HybridBody to hold the construction elements
construction_elements = hybrid_bodies.add()
construction_elements.name = "construction_elements"

# Create spline from arbitrary points

spline = hsf.add_new_spline()
spline.add_point(hsf.add_new_point_coord(L, 0, 0))
spline.add_point(hsf.add_new_point_coord(L/2, 0, L/8))
spline.add_point(hsf.add_new_point_coord(L/4, 0, L/2))
spline.add_point(hsf.add_new_point_coord(L/8, 0, L/2))
# Append spline
construction_elements.append_hybrid_shape(spline)
    
# Update the document
document.part.update()

# --------------------------------------------------------------------------- #
# Create surface revolution
# --------------------------------------------------------------------------- #

#  Get Z axis direction
Zdir = hsf.add_new_direction_by_coord(0, 0, 1)

surfRevolution = hsf.add_new_revol(spline, 0, 360, Zdir)

# Append surface
construction_elements.append_hybrid_shape(surfRevolution)

# Update the document
document.part.update()

# %% ------------------------------------------------------------------------ #
# Create an intersecting surface
# --------------------------------------------------------------------------- #

# Create polyline in Y axis

polyline = hsf.add_new_polyline()
polyline.insert_element(hsf.add_new_point_coord(0,-1.5*L, 0), 0)
polyline.insert_element(hsf.add_new_point_coord(0, 1.5*L, 0), 1)
polyline.closure = False

# Append polyline
construction_elements.append_hybrid_shape(polyline)
    
# Update the document
document.part.update()

#  Get Z axis direction
Zdir = hsf.add_new_direction_by_coord(0, 0, 1)

surfExtrusion = hsf.add_new_extrude(polyline, 1.5*L, 0, Zdir)
surfExtrusion.symmetrical_extension = True

# Append surface
construction_elements.append_hybrid_shape(surfExtrusion)

# Update the document
document.part.update()

# %% ------------------------------------------------------------------------ #
# Split revolution surface with planar surface
# --------------------------------------------------------------------------- #

split1 = hsf.add_new_hybrid_split(surfRevolution, surfExtrusion, -1) 
split1.both_sides_mode = False
split1.intersection_computation = True
construction_elements.append_hybrid_shape(split1)

# Update the document
document.part.update()

#  Note for this case that you cannot split the planar surface with the 
# revolution surface because it won't cut it completely: 
# hsf.add_new_hybrid_split(surfExtrusion, surfRevolution, 1) will lead to error 

# %% Hide elements

selection.clear();
selection.add(polyline)
selection.add(spline)
selection.add(surfRevolution)
selection.add(surfExtrusion)
selection.vis_properties.set_show(1) # 0: Show / 1: Hide
selection.clear()

Error:
Traceback (most recent call last):
  File "D:\pythonProject\surfaces\split.py", line 113, in <module>
    document.part.update()
  File "D:\pyCatia Module\pythonProject\.venv\Lib\site-packages\pycatia\mec_mod_interfaces\part.py", line 906, in update
    return self.part.Update()
           ^^^^^^^^^^^^^^^^^^
  File "<COMObject <unknown>>", line 2, in Update
pywintypes.com_error: (-2147352567, 'Exception occurred.', (0, 'CATIAPart', 'The method Update failed', None, 0, -2147467259), None)

**Desktop (please complete the following information):**
 - OS:  Windows 11
 - CATIA V5 (version V5-R20)
 - pycatia version (pycatia 08.0)

evereux, 2024-11-19:
Can you please provide some code that runs, trips up on the error you have reported - see below and is properly [formatted](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax) (see the section on formatting code). 

The traceback you supplied doesn't relate to the code you have provided. It trips up on line 113 but your code sample doesn't have that many lines?

FaiqShahbaz, 2024-11-19:
[split.txt](https://github.com/user-attachments/files/17812950/split.txt)
this is the exact code which i used, there are other codes as well in which i am experiencing the exact same error,

Traceback (most recent call last):
File "D:\pythonProject\surfaces\split.py", line 113, in
document.part.update()
File "D:\pyCatia Module\pythonProject.venv\Lib\site-packages\pycatia\mec_mod_interfaces\part.py", line 906, in update
return self.part.Update()
^^^^^^^^^^^^^^^^^^
File "<COMObject >", line 2, in Update
pywintypes.com_error: (-2147352567, 'Exception occurred.', (0, 'CATIAPart', 'The method Update failed', None, 0, -2147467259), None)

evereux, 2024-11-19:
``` python
polyline.insert_element(hsf.add_new_point_coord(0,-1.5*L, 0), 0)
```

The `polyline.insert_element` position should start at 1, not 0. 😃

In the future, if your script is breaking on a specific problem, in this case updating part after inserting the polyline, you should strip down your script to tackle that and that only. 

A polyline example

```python

from pycatia import catia
from pycatia.mec_mod_interfaces.part_document import PartDocument

caa = catia()
application = caa.application
document: PartDocument = application.active_document
part = document.part
hsf = part.hybrid_shape_factory
hybrid_bodies = part.hybrid_bodies
construction_geometry = hybrid_bodies.item('ConstructionGeometry')

point_1 = hsf.add_new_point_coord(0, 0, 0)
point_2 = hsf.add_new_point_coord(100, 100, 100)
construction_geometry.append_hybrid_shape(point_1)
construction_geometry.append_hybrid_shape(point_2)
part.update()

ref_point_1 = part.create_reference_from_object(point_1)
ref_point_2 = part.create_reference_from_object(point_2)

polyline = hsf.add_new_polyline()
polyline.insert_element(ref_point_1, 1)
polyline.insert_element(ref_point_2, 2)
polyline.closure = False

construction_geometry.append_hybrid_shape(polyline)

part.update()
```

maipj, 2024-12-27:
I also have some questions on my end regarding the update of the part. Below are some of my size updates, which may include parameter values such as size, offset, and boss. I have used the following methods to modify them all, but there have been errors. I would like to know how to modify all the parameters of CATIA or how to determine if they can be modified. If anyone knows, please let me know. I would be very grateful!

code down
        try:
            self.open_CATPart_file(cad_path)
            part_document: PartDocument = self.Catia_Instance.active_document
            part = part_document.part
            parameters = part.parameters
            for item in size_data:
                val = str_to_float(item[1])
                data = parameters.get_item_by_name(item[0])
                length = Length(data.com_object)
                length.value = val
                part.update()
        except Exception as ex:
            print("update failed", ex)

evereux, 2024-12-27:
Please take a look at the [examples](https://pycatia.readthedocs.io/en/latest/examples.html) regarding parameters, specifically: https://github.com/evereux/pycatia/blob/master/examples/example__parameters__002.py

If you still have a problem please open a new issue providing a complete code example demonstrating the problem you are having along with any error message you have received.

closed 2024-12-27 by evereux