evereux/pycatia #235: The selected feature doesn't belong to the current body

CesarAero, 2024-12-27:
Dear support,

I am trying to make the circular pattern of a hybrid shape (a loft in this case). I normally don't have problems when I try to do the pattern of an already existing solid part but I am struggling to make it work for a hybrid body with pycatia.

I have no issues when I do this with the GUI:
[image]

But it seems that there is a problem when I try to do the same with PyCATIA because the selected feature doesn't belong to the current body:
[image]

I cannot really understand where the error is coming from since the procedue is quite straight forward :(

I left the code in the following file:
[droneBlades.txt](https://github.com/user-attachments/files/18259299/droneBlades.txt)

Could the problem come from the way I've defined the hybrid bodies "construction_elements"? Should I add any internal references to include them into the actual body?

Thank you in advance!

evereux, 2024-12-27:
I can't run your code:

```
Traceback (most recent call last):
  File "C:\temp\pycatia\issue_235.py", line 264, in <module>
    document.part.update()
  File "C:\temp\pycatia\env\Lib\site-packages\pycatia\mec_mod_interfaces\part.py", line 906, in update
    return self.part.Update()
           ^^^^^^^^^^^^^^^^^^
  File "<COMObject <unknown>>", line 2, in Update
pywintypes.com_error: (-2147352567, 'Exception occurred.', (0, 'CATIAPart', 'The method Update failed', None, 0, -2147467259), None)
```

There are several other errors that also need fixing. Also, I had to install matplotlib, scipy and numpy (in a new env so I don't pollute my own) just to get that far. Would be much easier for me to have had a simplified version just dealing specifically with the problem, which seems to simply be creating a circular pattern?

> I have no issues when I do this with the GUI:

With my CATIA V5 set-up I can't create a circular pattern of a joined surface within a PartBody. I can't even then move the result to a PartBody.

Anyways ... I solved this by recording a macro and used that as a reference.

```python

# This example requires an open Part. 
# The Part shall have a Geometrical Set called ConstructionGeometry. 
# In that set there shall be a line `Line.1` which is used for the center of rotation 
# and a joined object `Joined.1` which is the item to be rotated.

from pycatia import catia
from pycatia.mec_mod_interfaces.part_document import PartDocument
from pycatia.enumeration.enumeration_types import cat_circular_pattern_parameters

caa = catia()
part_document: PartDocument = caa.active_document
part = part_document.part
sf = part.shape_factory
hbs = part.hybrid_bodies
hb_construction_geometry = hbs.item("ConstructionGeometry")
hs_construction_geometry = hb_construction_geometry.hybrid_shapes

joined_surface = hs_construction_geometry.item("Join.1")
direction = hs_construction_geometry.item("Line.1")
ref_direction = part.create_reference_from_object(direction)

reference_1 = part.create_reference_from_name("")
reference_2 = part.create_reference_from_name("")

circular_pattern = sf.add_new_surfacic_circ_pattern(
    joined_surface,
    1,
    2,
    20.0,
    45.0,
    1,
    1,
    reference_1,
    reference_2,
    True,
    0.0,
    True,
    False

)

circular_pattern.circular_pattern_parameters = cat_circular_pattern_parameters.index("catInstancesandAngularSpacing")
circular_pattern.set_rotation_axis(ref_direction)

part.update()
```

evereux, 2024-12-27:
So ultimately the problem was using [ShapeFactory.add_new_circ_pattern()](https://github.com/evereux/pycatia/blob/3418debfb1bd601c61a87d9556975fa50b6eaddb/pycatia/part_interfaces/shape_factory.py#L340). The correct method to use is [ShapeFactory.add_new_surfacic_circ_pattern()](https://github.com/evereux/pycatia/blob/3418debfb1bd601c61a87d9556975fa50b6eaddb/pycatia/part_interfaces/shape_factory.py#L2450) which works with the GSD objects.

Recording macros and seeing how CATIA V5 does it answers a lot of common questions.

CesarAero, 2024-12-30:
I see, I was using another code as a template where I did a circular pattern of a solid part using add_new_circ_pattern. But I just realized that even though the GUI has the same buttom for doing the pattern of parts of hybrid bodies, in the scripts you need to call a different function. Problem solved! Feel free yo close the thread :)

closed 2025-01-14 by evereux