evereux/pycatia #237: The type of product is CATpart. How to convert it to part?

lxy-mini, 2025-01-13:


lxy-mini, 2025-01-13:
The type of product is CATpart，how to save it as .CATpart? can you share the code ?

evereux, 2025-01-13:
Have you taken a look at the [examples](https://pycatia.readthedocs.io/en/latest/examples.html)? The section on Documents is probably what you want.

Failing that, can you please provide more information about what it is you're trying to do as the question doesn't make a lot of sense to me.

lxy-mini, 2025-01-14:
Okay，thanks！
this is my code
    if item.is_catproduct():
        product = item
        print(f'This is a product: "{item}"')

        if item.has_children():
            print("This product has children.")
            children = item.get_children()
            print(children)
            for child in children:
                print(type(child))
                if child.is_catpart():
                    # part = child.part
                    print(f'This is a part: "{child.name}"')
                    print_properties(child)
                    print(child.type)
    now! child.type is CATpart i want to save it as a new document which is .CATpart

evereux, 2025-01-14:
I'm still confused but anyways ...

> The type of product is CATpart，how to save it as .CATpart? can you share the code ?

I guess you mean here `Save As` as from your descriptions here it seems like it's already a CATPart. To save things you need a `Document` object, be it a `ProductDocument` / `PartDocument` etc. If you have a `Product` you can use it's `file_name` to get the `Document` object from the `Documents` collection.

```python

from pycatia import catia
from pycatia.enumeration.enumeration_types import cat_work_mode_type
from pycatia.mec_mod_interfaces.part_document import PartDocument
from pycatia.product_structure_interfaces.product_document import ProductDocument

caa = catia()
application = caa.application
documents = application.documents
product_document: ProductDocument = application.active_document
product = product_document.product
products = product.products
product.activate_terminal_node(products)  # might be needed, depends on CATIA configuration.

for _product in products:
    _product.apply_work_mode(cat_work_mode_type.index("DESIGN_MODE"))
    if _product.file_name.endswith('CATPart'):
        _part_document: PartDocument = documents.item(_product.file_name)
```

lxy-mini, 2025-01-15:
thanks a lot! 
This is the tree structure of my pycatia. first, this is a .stp document.
--A
 --B（product）
  --C1 （product）
   --D1 (part)
   --D2 (part)
  --C2 (part)
Now I read the entire document according to product，
`product = documents.product
`
I use the get_children() function in product to handle to D1, D1.type is a CATpart. Unfortunately, D1 is still a PRODUCT! I want D1 to be a part, so that I can call the part's functions for processing

```
from pycatia import catia
from pycatia.enumeration.enumeration_types import cat_work_mode_type
from pycatia.product_structure_interfaces.product_document import ProductDocument
from pycatia.product_structure_interfaces.product import Product
from pathlib import Path
from pycatia.in_interfaces.document import Document

def print_properties(obj: Product):
    print(
        f"{obj.name}: mass: {obj.analyze.mass}, \n"
        f"    volume: {obj.analyze.volume}, \n"
        f"    wet_area: {obj.analyze.wet_area}, \n"
        f"    gravity_center: {obj.analyze.get_gravity_center()}, \n"
        f"    inertia: {obj.analyze.get_inertia()}"
    )

caa1 = catia()
documents = caa1.active_document
product = documents.product
product.apply_work_mode(cat_work_mode_type.index("DESIGN_MODE"))

products = product.products
filename = ["xxx"]
if len(products) == 0:
    print("Active document has no children or is not a CATProduct.")

for item in products:
    print(item.type)
    if item.is_catpart():
        print(f'This is a part: "{item}"')
        print("")

    if item.is_catproduct():
        product = item
        print(f'This is a product: "{item}"')

        if item.has_children():
            print("This product has children.")
            children = item.get_children()
            print(children)
            for child in children:
                print(type(child))
                if child.is_catpart():

                    print(f'This is a part: "{child.name}"')
                    print_properties(child)
                    print(child.type)
                    print(child.position.get_components())

                elif child.has_children():
                    print(type(child))
                    oo = child.products
                    print(f'00"{oo}"')
                    for item in oo:
                        if item.is_catproduct():
                            print(type(item))
                            if item.has_children():
                                ggren = item.get_children()
                                for gg in ggren:
                                    print(f'gg.has_children"{(gg.has_children == True)}"')
                                    print(gg.part_number)

print("")

```

lxy-mini, 2025-01-15:
> I'm still confused but anyways ...
>
>
> I guess you mean here `Save As` as from your descriptions here it seems like it's already a CATPart. To save things you need a `Document` object, be it a [...]
yes!!! This is the answer I want!!! Thanks again to the author!!!

closed 2025-01-15 by lxy-mini