evereux/pycatia #239: a part consists of multiple MANIFOLD_SOLID_BREP. How do I split each MANIFOLD_SOLID_BREP into an independent part and save it?

lxy-mini, 2025-01-16:


evereux, 2025-01-16:
How would you do this manually? Can you reproduce this programmatically? What have you tried?

Have you taken the time to read: https://github.com/evereux/pycatia/issues/28

lxy-mini, 2025-01-16:
part = document.part
bodies = part.bodies
    for body in bodies:
        new_part_document = part.application.documents.add("Part")
        new_part = Part(new_part_document.part.com_object)
Sorry, as a beginner, I am not very familiar with the API functions.I tried to save the xyz plane and MANIFOLD_SOLID_BREP information in the original part to the newly created part, but no relevant functions were found. thanks for your help

evereux, 2025-01-16:
My immediate thoughts for solving this are:

1. Create a copy of the entire part then delete the items you __don't__ want?
2. Use the `Selection` object to copy selections from one part to another? Not sure if this is a thing since the `Selection` object is unique to the `Part`. Worth a try though?

For me, option 1 would seem to be the simplest.

lxy-mini, 2025-01-16:
okay， very good！I have been thinking about how to operate part

closed 2025-01-16 by lxy-mini