evereux/pycatia #243: Use a relation law to create a parallel surface

CesarAero, 2025-01-17:
Dear support,

I need to use a relation law to offset a line based on a geometrical relation which depends on parameters. I am facing 2 issues:

1) The defined relation law presents a yellow marker and therefore cannot be evaluated unless I manually open it and click "OK". Maybe there is an update of the part relations that I am not aware of.

2) I cannot pass this relation law as input for my add_new_curve_par() function.

I have uploaded a reproducer to show the issue:

`from pycatia import catia
from pycatia.mec_mod_interfaces.part_document import PartDocument
  
caa = catia()
application = caa.application
documents = application.documents
if documents.count > 0:
    for document in documents:
        document.close()
       
# Create Part
documents.add('Part')

# Get references to document, parts and tools to create geometries
document: PartDocument = application.active_document
part = document.part
partbody = part.bodies[0]
hybrid_bodies = part.hybrid_bodies
hsf = part.hybrid_shape_factory

# Create a new hybrid body
constElem = hybrid_bodies.add()
constElem.name = "Construction elements"

# Create a line
line = hsf.add_new_line_pt_pt(hsf.add_new_point_coord(0,0,0),hsf.add_new_point_coord(1000,0,0))
line.name = "line"
constElem.append_hybrid_shape(line)
part.update()

# Create parameters and relation law
scale_par = part.parameters.create_real('scale', 2)
scaleLaw = part.relations.create_law('z_shit', '', 'z = scale*xi')
scaleLaw.add_formal_parameter('z', 'LENGTH')
scaleLaw.add_formal_parameter('xi', 'Real')
scaleLaw.activate()
part.update()

# Create parallel curve with constant offset - Works with constant offset
shift_1 = hsf.add_new_curve_par(line, part.origin_elements.plane_xy, 50, False, False)
shift_1.name = "Constant offset"
constElem.append_hybrid_shape(shift_1)
document.part.update()

# Create parallel curve with law - DOESN'T WORK!
shift_2 = hsf.add_new_curve_par(line, part.origin_elements.plane_xy, 0, False, False)
shift_2.name = "Shift with relation law"
shift_2.law_type = 4
shift_2.curve_par_law = scaleLaw # <--- The code crashes here
constElem.append_hybrid_shape(shift_2)
document.part.update()`

If I do the last step by hand it works perfectly. Which can be the issue with my code? Thank you in advance for your time!

evereux, 2025-01-18:
I edited your post to make your code snippet more readable. :-)

[HybridShapeCurvePar.curve_par_law()](https://pycatia.readthedocs.io/en/latest/api/pycatia/hybrid_shape_interfaces/hybrid_shape_curve_par.html) takes a `Reference` object. Create a `Reference` from the `scaleLaw` object and it'll solve that crash.

Are you using an IDE with syntax highlighting? I'm using pycharm and the type checker flagged this.

Quite often the Class methods expect a `Reference` object passed to them, not the object itself. However, sometimes we can get away with not doing that as CATIA seems to do the right thing despite the API saying otherwise. 

For example: your usage of [HybridShapeFactory.add_new_curve_par](https://pycatia.readthedocs.io/en/latest/api/pycatia/hybrid_shape_interfaces/hybrid_shape_factory.html#pycatia.hybrid_shape_interfaces.hybrid_shape_factory.HybridShapeFactory.add_new_curve_par()). This expects reference objects for `curve` and `support`. You hadn't done this and it still worked.

I've tweaked your code to now do this:

```python

from pycatia import catia
from pycatia.mec_mod_interfaces.part_document import PartDocument

caa = catia()
application = caa.application
documents = application.documents
if documents.count > 0:
    for document in documents:
        document.close()

# Create Part
documents.add('Part')

# Get references to document, parts and tools to create geometries
document: PartDocument = application.active_document
part = document.part
partbody = part.bodies[0]
hybrid_bodies = part.hybrid_bodies
hsf = part.hybrid_shape_factory

# Create a new hybrid body
constElem = hybrid_bodies.add()
constElem.name = "Construction elements"

# Create a line
point_1 = hsf.add_new_point_coord(0, 0, 0)
point_2 = hsf.add_new_point_coord(1000, 0, 0)
ref_point_1 = part.create_reference_from_object(point_1)
ref_point_2 = part.create_reference_from_object(point_2)
line = hsf.add_new_line_pt_pt(ref_point_1, ref_point_2)
line.name = "line"
constElem.append_hybrid_shape(line)
ref_line = part.create_reference_from_object(line)
part.update()

# Create parameters and relation law
scale_par = part.parameters.create_real('scale', 2)
scaleLaw = part.relations.create_law('z_shift', '', 'z = scale*xi')
scaleLaw.add_formal_parameter('z', 'LENGTH')
scaleLaw.add_formal_parameter('xi', 'Real')
scaleLaw.activate()
ref_scaleLaw = part.create_reference_from_object(scaleLaw)
part.update()

# Create parallel curve with constant offset - Works with constant offset
ref_plane_xy = part.create_reference_from_object(part.origin_elements.plane_xy)
shift_1 = hsf.add_new_curve_par(ref_line, ref_plane_xy, 50, False, False)
shift_1.name = "Constant offset"
constElem.append_hybrid_shape(shift_1)
document.part.update()

# Create parallel curve with law - DOESN'T WORK!
shift_2 = hsf.add_new_curve_par(ref_line, ref_plane_xy, 0, False, False)
shift_2.name = "Shift with relation law"
shift_2.law_type = 4
shift_2.curve_par_law = ref_scaleLaw  # <--- The code crashes here
constElem.append_hybrid_shape(shift_2)
document.part.update()
```

We have an additional problem and that's with your `Law` creation.

```python

from pycatia import catia
from pycatia.mec_mod_interfaces.part_document import PartDocument

caa = catia()
application = caa.application
documents = application.documents

# Create Part
documents.add('Part')

# Get references to document, parts and tools to create geometries
document: PartDocument = application.active_document
part = document.part

# Create parameters and relation law
scale_par = part.parameters.create_real('scale', 2)
scaleLaw = part.relations.create_law('z_shift', '', 'z = scale*xi')
scaleLaw.add_formal_parameter('z', 'LENGTH')
scaleLaw.add_formal_parameter('xi', 'Real')
scaleLaw.activate()
part.update()

```

I don't know the reason for this. Can you record a macro doing just this step and post that here please.

evereux, 2025-01-18:
I think the problem was creating the `Law` referencing `Paramater`(s) that don't yet exist. 

I'm geussing but when the `Law` is created CATIA V5 tries to evaluate it straight away, which it can't.  There doesn't seem to be a method to get it to re-evaluate once the referenced parameters are created.

Anyways, here's the solution:

```python
from pycatia import catia
from pycatia.mec_mod_interfaces.part_document import PartDocument

caa = catia()
application = caa.application
documents = application.documents

# Create Part
documents.add('Part')

# Get references to document, parts and tools to create geometries
document: PartDocument = application.active_document
part = document.part
parameters = part.parameters
relations = part.relations

# Create parameters and relation law
scale_par = parameters.create_real('scale', 2)
l = '`z` = `scale`*`xi`'
scaleLaw = relations.create_law('z_shift', '', '')
scaleLaw.add_formal_parameter('z', 'Length')
scaleLaw.add_formal_parameter('xi', 'Real')
scaleLaw.modify(l)
```

You don't need to surround the parameter names in back ticks for the law string but it'd be good practice to do so (using parameters with spaces in names etc).

CesarAero, 2025-01-18:
It works now! The problem was indeed to set the formula too early before defining the formal internal parameters. Thanks a lot for your help!

evereux, 2025-01-18:
Glad it's working for you now. Think I'll add this to the examples.

CesarAero, 2025-01-18:
Definitely, it would be super helpful to someone running into this issue too!

closed 2025-02-04 by CesarAero