evereux/pycatia #244: [FEATURE REQUEST]

ozserhatt, 2025-01-17:
Hello dear friends !

How can I know the filter names for catia objects ? 
This is the example for it : If we can choose an AxisSystem, it called "AxisSystem" but
if I wanna choose a curve in geometrical set Where I can learn what can I say ?

```python
selection = part_document.selection
selection.clear()
filter_ = ('AxisSystem',)
logger.info('Select the Axis system.')
caa.message_box('Please select the Bounding Box Axis System.', buttons=0, title='Selection Prompt')
output = selection.select_element2(filter_, 'Select bounding box AxisSystem.', False)
if output == 'Cancel':
    logger.info('Exiting script.')
    exit()
```

evereux, 2025-01-18:
I don't think the selection object can do filterting based on geometrical set names.

Have a play with the Search feature in CATIA V5 itself. If you can solve it there you can use the output of that to help define your search criteria.

evereux, 2025-02-10:
Closing issue as the question is a little too broad to be supported here. Please refer to CATIA V5 documentation and other online resources regarding `Selection` filtering.

closed 2025-02-10 by evereux