evereux/pycatia #245: Judgment of intersection of two parts.

lxy-mini, 2025-01-20:
I want to determine whether the two parts are included in each other by judging the intersection relationship. But encountered some problems

```
from pycatia import catia
from pycatia.mec_mod_interfaces.part_document import PartDocument
from pycatia.product_structure_interfaces.product_document import ProductDocument
from pycatia.product_structure_interfaces.product import Product

# 初始化 CATIA
caa1 = catia()
application = caa1.application
documents = application.documents
product_document: ProductDocument = application.active_document

# 获取产品
product12 = product_document.product
product122 = product12.products

# 获取子产品（PartA 和 PartB）
sub1_product: Product = product122.item(1)  # PartA
sub2_product: Product = product122.item(2)  # PartB

# 获取 PartA 和 PartB 的零件文档
_part_document: PartDocument = documents.item(sub1_product.file_name)
partA = _part_document.part
_part_document: PartDocument = documents.item(sub2_product.file_name)
partB = _part_document.part

# 获取 PartA 和 PartB 的几何体
body_A = partA.bodies.item(1)  # 假设 PartA 的主几何体是第一个
body_B = partB.bodies.item(1)  # 假设 PartB 的主几何体是第一个

# 设置当前工作对象为 body_A
partB.in_work_object = body_B

# 进行布尔相交运算
intersect_result = partB.shape_factory.add_new_intersect(body_A)
partB.update()
```
Code execution error
```
Traceback (most recent call last):
  File "C:\code\Automodel\zzz.py", line 38, in <module>
    partB.update()
  File "C:\Users\.conda\envs\py39\lib\site-packages\pycatia\mec_mod_interfaces\part.py", line 906, in update
    return self.part.Update()
  File "<COMObject <unknown>>", line 2, in Update
pywintypes.com_error: (-2147352567, '发生意外。', (0, 'CATIAPart', '方法 Update 失败', None, 0, -2147467259), None)
```

evereux, 2025-01-20:
I suspect the issue here is trying the boolean `Intersect` on a `Body` that doesn't yet exist in the part. 

This isn't something you could do manually. Copy the source `Body` into the target `Part` first then use that to intersect.

lxy-mini, 2025-01-20:
okay，thank you！
partA and partB come from two different coordinate systems. I used coordinate transformation to put them in the same coordinate system.Theoretically, partA should be located inside partB. If you follow this operation （Copy the source `Body `into the target `Part `first then use that to intersect.）, will there be any conflict with coordinate conversion?

evereux, 2025-01-20:
I don't know, I can't see what you're actually doing. Try it?

closed 2025-02-04 by evereux