evereux/pycatia #246: boundary of the body

lxy-mini, 2025-01-23:
I want to get the boundary of the body. Can you give me some ideas? I want to get the boundary of the body to calculate whether it is contained by other parts. I saw related functions to obtain the boundary, but a label is required. Is the label here pre-generated? 

```
   def get_boundary(self, i_label: str) -> Boundary:
        """
        .. note::
            :class: toggle

            CAA V5 Visual Basic Help (2020-07-06 14:02:20.222384))
                | o Func GetBoundary(CATBSTR iLabel) As Boundary
                | 
                |     Returns a boundary using its label.
                | 
                |     Parameters:
                | 
                |         iLabel
                |             Identification of the 
                | 
                |         Boundary object. See Reference.DisplayName. 
                |     Returns:
                |         The retrieved boundary

        :param str i_label:
        :rtype: Boundary
        """
        return Boundary(self.shapes.GetBoundary(i_label))
```

lxy-mini, 2025-01-23:
In addition, the body has been encapsulated and no more details can be seen in CATIA.

evereux, 2025-02-04:
I'm not familiar with the Shapes workbench at all I'm afraid. 

I don't think the boundary option here will do what you're hoping.

closed 2025-02-08 by lxy-mini