evereux/pycatia #247: Machining and PPR

miguelcaralb, 2025-01-28:
Hello, I am trying to access the tools and operations of my manufacturing program to perform a bulk export of the parameters from my programs. The problem is that I cannot properly define the PPR to access the activities and iterate through them. No matter how I do it, I always encounter the error that 'ProcessDocument' object has no attribute 'processes', even though I am declaring it as a PPR document. Please help.

`from pycatia import catia
from pycatia.ppr_interfaces.ppr_document import PPRDocument
from pycatia.dmaps_interfaces.process_document import ProcessDocument
from pycatia.dmaps_interfaces.activity import Activity
from pycatia.manufacturing_interfaces.machining_process import MachiningProcess
from pycatia.manufacturing_interfaces.manufacturing_setup import ManufacturingSetup
from pycatia.in_interfaces.document import Document

caa = catia()
#process_document = ProcessDocument(document.com_object)
process_document: ProcessDocument = caa.active_document
print(process_document.name)
#ppr = process_document.ppr_document
ppr: PPRDocument = caa.active_document
processes = ppr.processes
try:
    print(processes)
except Exception as e:
    print(e)`

Thank you very much for your time. Best regards.

evereux, 2025-01-29:
I don't know anything about the workbenches you mention but could you clarify what your `active_document` actually is? `ProcessDocument` or `PPRDocument`?  Doing `caa.active_document` for both types doesn't make much sense. 

Also, can you post a screenshot of a very simple Document tree so I may try and reproduce it.

evereux, 2025-01-29:
OK, so I had a quick play. Take a look at this:

```python

from pycatia import catia
from pycatia.dmaps_interfaces.process_document import ProcessDocument
from pycatia.ppr_interfaces.ppr_document import PPRDocument

caa = catia()
# process_document = ProcessDocument(document.com_object)
process_document: ProcessDocument = caa.active_document

ppr_document = process_document.ppr_document
ppr_document = PPRDocument(ppr_document.com_object)
processes = ppr_document.processes
```

`ProcessDocument.ppr_document()` actually returns an `AnyObject` which is in accordance with the docs. But I think this should be a `PPRDocument`. I'll patch pycatia in the coming days so we don't need to do this extra step `ppr_document = PPRDocument(ppr_document.com_object)`.

miguelcaralb, 2025-01-29:
First of all, thank you very much for your prompt response. I really appreciate your help with the issue I was facing.

I am attaching an image to better explain what I am trying to do. The goal is to iterate through the ManufacturingProgram that exists in each Part Operation.1. Then, I want to go through the operations contained within each tool and export the FeedRate and SpindleSpeed parameters.

[image]

I am still far from achieving it, but with your help, I was able to overcome the first obstacle. Thank you again.

Best regards.

closed 2025-02-04 by evereux