evereux/pycatia #248: Use of add_new_circle_center_tangent

CesarAero, 2025-01-29:
Dear support,

I am trying to create a circle centered in a point which is tangent to a curve. I am using the hybrid shape factory command "add_new_circle_center_tangent" as:

myCircle = hsf.add_new_circle_center_tangent(pt_ref, curve_ref, plane_ref, 10)

The references work well but my problems comes from the fact that I have to provide a radius (10 in this case) which is irrelevant since there is only 1 circle solution for this case. This causes a warning in CATIA that I can only fix by manually entering in the operation and pressing "ok", then the GUI does its stuff and fixes the issue... How can I fix this in my code to ignore the radius entry?

Thank you in advance for your time!

evereux, 2025-01-29:
I can't solve your issue at the moment but the question I would ask myself is this ... how would this be handled in VBA / CATScript?

CesarAero, 2025-01-29:
So I've build a small reproducer to the problem:

# ----- Packages
# --------------------------------------------------------------------------- #
from pycatia import catia
from pycatia.mec_mod_interfaces.part_document import PartDocument

# ----- Set CATIA environment
# --------------------------------------------------------------------------- #
caa = catia()
application = caa.application
documents = application.documents
if documents.count > 0:
    for document in documents:
        document.close()
        
# Create Part
documents.add('Part')

# Get references to document, parts and tools to create geometries
document: PartDocument = application.active_document
part = document.part
partbody = part.bodies[0]
hybrid_bodies = part.hybrid_bodies
hsf = part.hybrid_shape_factory

# Set PartBody as a working object
part.in_work_object = partbody

# ----- Perform operations
# --------------------------------------------------------------------------- #

# Arbitrary dimension
L = 100

construction_elements = hybrid_bodies.add()
construction_elements.name = "construction_elements"

p_0 = hsf.add_new_point_coord(0, 0, 0)
p_0.name = "p_0"
construction_elements.append_hybrid_shape(p_0)
document.part.update()

p_1 = hsf.add_new_point_coord(L, -L, 0)
p_1.name = "p_1"
construction_elements.append_hybrid_shape(p_1)
document.part.update()

p_2 = hsf.add_new_point_coord(L, L, 0)
p_2.name = "p_2"
construction_elements.append_hybrid_shape(p_2)
document.part.update()

lineObject = hsf.add_new_line_pt_pt(p_1,p_2)
lineObject.name = "lineObject"
construction_elements.append_hybrid_shape(lineObject)

# Update the document
document.part.update()

# ----- Create tangent circle
# --------------------------------------------------------------------------- #
tangentCircle = hsf.add_new_circle_center_tangent(p_0, lineObject, part.origin_elements.plane_xy, 10)
tangentCircle.name = "tangentCircle"
construction_elements.append_hybrid_shape(tangentCircle)
document.part.update()

So the problem comes after the last update where there is a warning in my case and the operation fails.

The equivalent process in CATIA macros would be the following (after creating the previous constructions bodies):

Set hybridShapePointCoord1 = hybridShapes1.Item("p_0")

Set reference1 = part1.CreateReferenceFromObject(hybridShapePointCoord1)

Set hybridShapeLinePtPt1 = hybridShapes1.Item("lineObject")

Set reference2 = part1.CreateReferenceFromObject(hybridShapeLinePtPt1)

Set originElements1 = part1.OriginElements

Set hybridShapePlaneExplicit1 = originElements1.PlaneXY

Set reference3 = part1.CreateReferenceFromObject(hybridShapePlaneExplicit1)

Set hybridShapeCircleCenterTangent1 = hybridShapeFactory1.AddNewCircleCenterTangent(reference1, reference2, reference3, 20.000000)

hybridShapeCircleCenterTangent1.DiscriminationIndex = 1

hybridShapeCircleCenterTangent1.BeginOfCircle = 0

hybridShapeCircleCenterTangent1.Orientation1 = 1

hybridShapeCircleCenterTangent1.Orientation2 = 1

hybridShapeCircleCenterTangent1.TangentOrientation1 = 1

hybridShapeCircleCenterTangent1.TangentOrientation2 = 1

hybridShapeCircleCenterTangent1.SetLimitation 1

hybridBody1.AppendHybridShape hybridShapeCircleCenterTangent1

part1.InWorkObject = hybridShapeCircleCenterTangent1

part1.Update 

I also tried setting the other attributes but it doesn't work as expected...

evereux, 2025-01-30:
Well, this is very curious. I have run your python script and executes just fine without any warning. 😕 I can only test in R21 for now.

@Mithro86, @deloarts, @Djey51: if convenient could any of you try running @CesarAero 's python script above please? Thanks!

Djey51, 2025-01-30:
Hello @evereux  and @CesarAero ,

I tested your script without warning message on V5-62015 release with MD2 license.

Did you try from default DS settings?

Best regards,

CesarAero, 2025-01-30:
Actually I thought that the problem could come from my computer at work (the reproducer works in my local machine) but today I tested the reproducer that I sent in my previous post and it works now... So the problem might come from another side. I have take some snaps to show the issue.

Here you can see a point and a spline which are contained on the same plane/support. After running my code (same lines as reproducer) I have the following warning:

[image]

But if I go inside the operation and press "ok", everything is fine (well the circle doesn't look tangent due to the low teselation level):

[image]

Is the CATIA GUI doing something extra (like discarding solutions) that we should specify when coding?

Djey51, 2025-01-31:
Hi @CesarAero,

When you have the warning.
Go to the TangentCircle, right mouse button\local update. You should obtain a diagnostic windows popup.
Could you share us this windows and its full content please?

Best regards

CesarAero, 2025-01-31:
This is the diagnosis:
```
Feature cannot be computed.
The error may be due to :
 - no possible solution or
 - an input orientation change or
 - inputs relative positionning change.
Edit the feature and choose a correct solution.
```

It seems that I have to test a bit with the operation parameters (discrimination_index,begin_of_circle,orientation1,orientation2, etc) but I cannot find the right combination.

CesarAero, 2025-02-04:
Ok by doing several tests I found the right combination to create the tangent circle :)

closed 2025-02-04 by CesarAero