evereux/pycatia #249: USER DEFINITION FEATURE !!! UDF , How can i find from knowledge_templage to UserFeature

ozserhatt, 2025-02-03:
Hello dear frineds  and dear evereux !!!

How can I create a UserFeature basically ? I can not in library. Can you reference to me ?

evereux, 2025-02-03:
I can't find anything online that details information about the creation of a `User Feature`.

I did find this about instantiating a [ `User Feature` from a macro](http://catiadoc.free.fr/online/cfyugpkt_C2/cfyugpktmt0001.htm) but it doesn't help with the creation.

evereux, 2025-02-03:
Just to add recording a macro creating a User Feature didn't reveal anything.

ozserhatt, 2025-02-03:
I will perform it, but just i can not reach class of shape_instance and instance_factory.
Pycatia library has both class.

```python
def __init__(self):
     self.caa = catia()
     self.application = self.caa.application
     self.document = self.application.active_document
     self.part = Part(self.document.part.com_object)
     self.shape_instance = self.part.-----------
``` 

If I can reach the class I can do it for UserFeature. I can share it.
Can you help me how can I reach these classes.

ozserhatt, 2025-02-03:
You need to create first UserFeature with class of ShapeInstance. You can use these functions to perform your first step.
The second step will be with class of InstanceFactory functions.

ozserhatt, 2025-02-03:
For the first one I need to create userfeature, and then I will create an instance.
By the way, thank you so much for reaching classes.

ozserhatt, 2025-02-03:
Sorry !!! I can not reach the class of ShapeInstance. It is not reachable with get_customer_factory.
I have an error like this :
```python
File "<COMObject <unknown>>", line 2, in GetCustomerFactory
pywintypes.com_error: (-2147352567, 'Özel durum oluştu.', (0, 'CATIAPart', 'The method GetCustomerFactory failed', None, 0, -2147467259), None)
```

evereux, 2025-02-03:
Do you have any examples of doing this in VBA / CATScript? I don't know what should be passed to ShapeInstance to initialise it.

evereux, 2025-02-03:
That's python, not VBA / CATScript.

I'm not convinced that what you're trying to do can be done.

evereux, 2025-02-03:
That doesn't help me.

> Do you have any examples of doing this in VBA / CATScript? I don't know what should be passed to ShapeInstance to initialise it.

> I'm not convinced that what you're trying to do can be done.

ozserhatt, 2025-02-03:
This is the macro that i created. 
My instructions  is click the command of UserFeature like ;

[image]

After choosing some created objects and finish. It gave me a vb code is : 

```
Language="VBSCRIPT"

Sub CATMain()

Set partDocument1 = CATIA.ActiveDocument

Set part1 = partDocument1.Part

Set parameters1 = part1.Parameters

Set angle1 = parameters1.Item("Create-Casting-Steel\TRIM-STEEL-7\inputs\Join.58\Angular threshold")

Dim boolean1
boolean1 = angle1.IsTrueParameter

Set parameters2 = part1.Parameters

Set length1 = parameters2.Item("Create-Casting-Steel\TRIM-STEEL-7\inputs\Join.58\Merging distance")

Dim boolean2
boolean2 = length1.IsTrueParameter

Set parameters3 = part1.Parameters

Set angle2 = parameters3.Item("Create-Casting-Steel\TRIM-STEEL-7\inputs\Join.57\Angular threshold")

Dim boolean3
boolean3 = angle2.IsTrueParameter

Set parameters4 = part1.Parameters

Set length2 = parameters4.Item("Create-Casting-Steel\TRIM-STEEL-7\inputs\Join.57\Merging distance")

Dim boolean4
boolean4 = length2.IsTrueParameter

part1.Update 

End Sub

```

ozserhatt, 2025-02-03:
Actually class of the ShapeInstance is creating by class Shape. May be relate with this ? Because other class has AnyObject.

```python
class ShapeInstance(Shape):

    """
        .. note::
            :class: toggle

            CAA V5 Visual Basic Help (2020-06-11 12:40:47.360445)

                | System.IUnknown
                |     System.IDispatch
                |         System.CATBaseUnknown
                |             System.CATBaseDispatch
                |                 System.AnyObject
                |                     MecModInterfaces.Shape
                |                         ShapeInstance
                | 
                | The interface to access a CATIAShapeInstance.
```

ozserhatt, 2025-02-03:
Problem is solved !!! I reach the class of ShapeInstance.

```python
_shape = Shape(self.part.com_object)
self.shape_instance = ShapeInstance(_shape.com_object)
print(self.shape_instance)
```

ozserhatt, 2025-02-04:
Hi dear evereux ! Please can you support me ? I can not solve this problem all night :(
I am going crazy. How can I use these methods ? ( ShapeInstance )

```python
    def user_definition_feature(self):

        body = self.bodies.item(config.final_body.name)
        shapes = body.shapes
        shapes_count = shapes.count
        print(shapes_count)

        _shape = body.shapes.item(1)
        shape_instance = ShapeInstance(_shape.com_object)
        print(shape_instance)

        input_count = shape_instance.inputs_count
        for i in range(input_count):
            print(f"Input {i}")
``` 

```python
File "C:\Users\serha\Desktop\myprojects\Design\.venv\lib\site-packages\pycatia\mec_mod_interfaces\shape_instance.py", line 60, in inputs_count
    return self.shape_instance.InputsCount
  File "C:\Users\serha\Desktop\myprojects\Design\.venv\lib\site-packages\win32com\client\dynamic.py", line 629, in __getattr__
    raise AttributeError(f"{self._username_}.{attr}")
AttributeError: Item.InputsCount
```

ozserhatt, 2025-02-05:
OK bro, still thank you so much !

evereux, 2025-02-10:
Closing.

See also #251.

closed 2025-02-10 by evereux