evereux/pycatia #250: Set "Update relation at global update command"

CesarAero, 2025-02-04:
Dear support,

I have a part with several relations which depend on parameters and some of these relations are measurements from the objects created as a function of these parameters. I found out that if there is a relation with a "measurement" I need to first update the part and then the relation unless I go to the relation properties and set "Update relation at global update command". Is it possible to set this property to "True" for all the relations/formulas available in the part?

I tried to use [knowledge_sheet_setting_att](https://pycatia.readthedocs.io/en/latest/api/pycatia/knowledge_interfaces/knowledge_sheet_setting_att.html#module-pycatia.knowledge_interfaces.knowledge_sheet_setting_att) because it seems that there are some update attributes but I was not successful.

Is it possible to set an automatic update of all the relations by using macros?

evereux, 2025-02-04:
> I tried to use [knowledge_sheet_setting_att](https://pycatia.readthedocs.io/en/latest/api/pycatia/knowledge_interfaces/knowledge_sheet_setting_att.html#module-pycatia.knowledge_interfaces.knowledge_sheet_setting_att) because it seems that there are some update attributes but I was not successful.

This should have worked.

```python
from pycatia import catia
from pycatia.knowledge_interfaces.knowledge_sheet_setting_att import KnowledgeSheetSettingAtt

caa = catia()
application = caa.application
setting_controllers = application.setting_controllers()

_kssa: KnowledgeSheetSettingAtt = setting_controllers.item("CATLieKnowledgeSheetSettingCtrl")
kssa = KnowledgeSheetSettingAtt(_kssa.com_object)
kssa.relations_update_in_part_context_evaluate_during_update = 1
```

Once done all newly created relations should have the flag set correctly.

Can't see a way to do this on an individual `Relation` basis though.

CesarAero, 2025-02-16:
I see, so these two lines control the CATIA configuration 
```
kssa.relations_update_in_part_context_synchronous_relations = 1
kssa.relations_update_in_part_context_evaluate_during_update = 1
```
So you can change the following options:

[image]

This will affect the future relations created, but it won't modify the already created. It is fine for me! Thanks for the reply!

closed 2025-02-16 by CesarAero

evereux, 2025-02-17:
Keep in mind that it's possible for CATIA V5 to be configured by the Admin to lock those settings. If this is a concern, maybe check that the value was actually changed.