evereux/pycatia #253: Something went wrong when fitting the plane in discrete points

lxy-mini, 2025-02-28:
`from pycatia import catia
from pycatia.mec_mod_interfaces.hybrid_shape import HybridShape
from pycatia.in_interfaces.reference import Reference
import numpy as np
catia = catia()
document = catia.documents.add("Part")
part = document.part
hybrid_bodies = part.hybrid_bodies
new_hybrid_body = hybrid_bodies.add()
new_hybrid_body.name = "Noisy_Plane"
plane_origin = [0, 0, 0]
plane_normal = [0, 0, 1]
plane_size = 100
sampling_step = 10
points = []
for x in np.arange(-plane_size / 2, plane_size / 2, sampling_step):
    for y in np.arange(-plane_size / 2, plane_size / 2, sampling_step):
        points.append([x, y, 0])
points = np.array(points)
mean = 0
std_dev = 2
noise = np.random.normal(mean, std_dev, points.shape)
noisy_points = points + noise
factory = part.hybrid_shape_factory
for point in noisy_points:
    new_point = factory.add_new_point_coord(point[0], point[1], point[2])
    new_hybrid_body.append_hybrid_shape(new_point)

fill_surface = factory.add_new_fill()
for point in new_hybrid_body.hybrid_shapes:
    fill_surface.add_bound(Reference(point.com_object))
new_hybrid_body.append_hybrid_shape(fill_surface)
part.update()

`
hello!
I tried to apply a Gaussian distribution to one plane and refit it to the plane, but there was a problem with part.update(). Please show me.

evereux, 2025-03-01:
You aren't creating a `Plane` here but a fill `Surface`. What you're trying to do here isn't going to work. 

Is it [HybridShapeFactory.add_new_plane_mean](https://pycatia.readthedocs.io/en/latest/api/pycatia/hybrid_shape_interfaces/hybrid_shape_factory.html#pycatia.hybrid_shape_interfaces.hybrid_shape_factory.HybridShapeFactory.add_new_plane_mean) that you really want?

evereux, 2025-03-16:
Closing as have no feedback from OP.

closed 2025-03-16 by evereux