evereux/pycatia #257: [BUG]

naveenklt, 2025-03-03:
**Describe the bug**
I am beginner and I am unable to connect to the active CATIA Application

**To Reproduce**
Connect to active Catia Application using caa = catia() line

**Expected behavior**
connect to active catia application

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: [Windows 11]
 - CATIA V5 version [ V5-6R2018]
 - Python version 3.10.9
 - pycatia version [pycatia 0.8.1]

I tried with VB.Net also but CATIA = GetObject(, "CATIA.APPLICATION") line create exception like "Cannot create ActiveX component."

evereux, 2025-03-04:
I suspect that this is a problem with the CATIA V5 installation itself. You can try reinstalling (no need to uninstall) or take a look at the following:

https://pycatia.readthedocs.io/en/latest/installation.html#registering-the-catia-v5-com-server

https://github.com/evereux/pycatia-tools/issues/2#issuecomment-2326844060 <- this thread is regarding using a different tool but the root issue I think is the same.

naveenklt, 2025-03-04:
It's working. Thanks for the support.

closed 2025-03-04 by naveenklt

evereux, 2025-03-04:
Great, glad you got it working.