evereux/pycatia #258: Advanced Machining Classes and Objects

hamlintyler94, 2025-03-18:
Evereux, 

looking for some insight on how to track down what the classes, objects, etc do inside the machining environment specifically. IS there a private way to contact you for a email / conference call? 

My company is a large machine shop, and would be happy to work in tandem with you to develop out the advanced Machining side of PYCatia :D

evereux, 2025-03-19:
Simply, pycatia is a wrapper written in python that theoretically* enables you to do whatever it is you can with VBA. So, if you have a good understanding of what can be achieved with VBA, it's the same.

I've only ever spent an hour or so working with the machining environment in 3D. I haven't looked at what the automation side has to offer. So any help I can offer will be limited other than saying read the source code. ;-) I could work it out but time, licenses etc are an issue.

Ultimately you should read whatever documentation is available from Dassault and elsewhere online to see what can be achieved.

I've too much on right now to take on any new projects, thanks. :-) (maybe in several months time ...)

* I haven't ever used the machining classes so I don't know what potential bugs there might be. they'll be easy to fix though if the problem is made clear and easy to debug.

evereux, 2025-03-19:
Oh and a top tip that I need to include in my replies:

Record a macro doing what it is you're trying to do and learn from there too. The code will be rubbish but will point you in the right direction. It's invaluable.

evereux, 2025-08-14:
Closed as no further feedback from OP.

closed 2025-08-14 by evereux