evereux/pycatia #260: [BUG] Bounding box computation

Morpheus5828, 2025-04-28:
Hello, 
I used the same code to generate the bbox:https://pycatia.readthedocs.io/en/developement/user_scripts.html#create-bounding-box
But when i would like to apply this to these figure: http://catiadoc.free.fr/online/ccvug_C2/ccvugbt1400.htm
Catia will draw this figure: 

[image]

And i don't know why.
I select Axis system 3 and then a touch a random part of my figure.

Best regars

Marius

evereux, 2025-04-28:
In the future, please provide the full error log that python outputs to aid debugging.

This problem I think stems from the fact that you're using a French locale (made known in email to me, it isn't mentioned here). 

When I change to a french locale I get:

```python
(env_312) PS C:\Users\evereux\cloud\pycatia> python .\user_scripts\create_bounding_box.py
[2025-04-28 14:00:29,692] INFO in create_bounding_box: Select the Axis system.
[2025-04-28 14:00:32,765] INFO in create_bounding_box: Select a face of the body.
Traceback (most recent call last):
  File "C:\Users\evereux\cloud\pycatia\user_scripts\create_bounding_box.py", line 194, in <module>
    axis_2d = geometric_elements_1.item('AbsoluteAxis')
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\evereux\cloud\pycatia\pycatia\mec_mod_interfaces\geometric_elements.py", line 81, in item
    return GeometricElement(self.geometric_elements.Item(i_index))
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<COMObject <unknown>>", line 2, in Item
pywintypes.com_error: (-2147352567, 'Exception occurred.', (0, 'CATIAGeometricElements', 'La méthode Item a échoué', None, 0, -2147352573), None)
```

Notice the error `axis_2d = geometric_elements_1.item('AbsoluteAxis')`? In the french locale, the sketch axis system is not called `AbsoluteAxis`.

You'll also have problems with the lines:

```python

line_hdirection = Geometry2D(axis_2d.get_item('HDirection').com_object)

line_vdirection = Geometry2D(axis_2d.get_item('VDirection').com_object)

geometry_zx = geometric_elements_zx.item('Mark.1')
```

You'll need to replace `HDirection`, `VDirection` and `Mark.1` with the french locale equivalents. I know the answer, but I'll leave that for you to figure out, ;-)

I'll update the source script to let users know to look out for this. I'll also include the answers there for you.

Morpheus5828, 2025-04-28:
Thanks for answering, 
I update Catia and use english version, then i steal have a trouble: on axis z, it's not centering

[image]

`"""
    Example - Inertia - 001:

    Catia must be running with a Part open that contains a solid object
        within the MainBody (PartBody).

    This example will only measure the inertia for the MainBody.
        Additional PartBodies will be ignored.

    Options for measure main body is not use:
    If you Granularity_mode sets to 1 for all bodies
    and interia=spa_i.add(body) change to interia=spa_i.add(part) then
    in this case you can granularity_mode sets to 1 for all bodies
    or granularity_mode sets to 0 for main body

    Read comment in script to change density or select another body.

"""

##########################################################
# insert syspath to project folder so examples can be run.
# for development purposes.
import os
import sys

sys.path.insert(0, os.path.abspath('..\\pycatia'))
##########################################################

from pycatia import catia
from pycatia.mec_mod_interfaces.axis_system import AxisSystem
from pycatia.cat_logger import create_logger
from pycatia.enumeration.enumeration_types import cat_constraint_type, cat_vis_property_show, cat_constraint_mode
from pycatia.knowledge_interfaces.length import Length
from pycatia.mec_mod_interfaces.axis_system import AxisSystem
from pycatia.mec_mod_interfaces.part_document import PartDocument
from pycatia.sketcher_interfaces.geometry_2D import Geometry2D
from tool import get_inertia_matrix

__author__ = '[ptm] by plm-forum.ru'
__status__ = 'alpha'

catpart_file_path = "C://Users//A609568//PycharmProjects//Processing//FilterCover.CATPart"

caa = catia()
documents = caa.application.documents
part_document: PartDocument = documents.open(catpart_file_path)
part = part_document.part
part.update()

bodies = part.bodies

body = part.main_body

body_by_name = bodies.get_item_by_name('PartBody')

cog = get_inertia_matrix(part, part_document)

################################

caa = catia()
logger = create_logger()
application = caa.application
part_document: PartDocument = application.active_document

if '.CATPart' not in part_document.name:
[... 303 lines cut ...]
selection.vis_properties.set_real_opacity(150, 1)
selection.vis_properties.set_real_width(2, 1)

part.update()

def get_inertia_matrix(part, part_document):
    body = part.main_body
    spa_workbench = part_document.spa_workbench()
    spa_i = spa_workbench.inertias
    inertia = Inertia(spa_i.add(body).com_object)
    cog = inertia.get_cog_position()

    print('-----------------\n'
          'Center Of Gravity\n'
          '-----------------\n'
          f'X={cog[0] * 1000}\n'
          f'Y={cog[1] * 1000}\n'
          f'Z={cog[2] * 1000}')

    return cog

`

I update your main code using Cog matrix, but it has same result
Best regars

evereux, 2025-04-28:
The original issue is resolved.

I can reproduce the error you're now having: that is the bounding box being offset from the main part. The bounding box is incorrectly using the axis system as it's base limit.

I'll take a look at that at some point this week hopefully. Bonus points if you figure it out yourself. 😉 

I've no idea why you've pasted all that code above for inertia though. Seems to be irrelevant to the problem 😕

evereux, 2025-05-17:
Sorry for the delay, hoped to get to this sooner.

The script [create_bounding_box.py](https://github.com/evereux/pycatia/blob/master/user_scripts/create_bounding_box.py) has now been updated to fix this error.

You can see what was done to fix this if you look at the diff above: [fixed bounding box creation for when axis system isn't directly on part](https://github.com/evereux/pycatia/commit/6c870ec1b231a692dbfed950400dc2f1ea7a59ee)](https://github.com/evereux/pycatia/commit/6c870ec1b231a692dbfed950400dc2f1ea7a59ee)

I made some additional tweaks regarding the bounding box offset as that wasn't quite handled correctly. Also, added explicit names for some elements to make debugging easier for future me. 😄 

Any issues let me know.