evereux/pycatia #261: [BUG] is product method in in_interfaces\document.py

Morpheus5828, 2025-04-30:
Hello, 
I would like to use method called is_product() in document.py file, but it doesn't seems work.
This is my code:

caa = catia()
application = caa.application
part_document: PartDocument = application.active_document
print(part_document.is_product())

[image]

And when i look the doc, i don't see self.product declaration ...

Best regards

Marius

evereux, 2025-04-30:
`Document.is_product()` requires a `ProductDocument` object. This isn't clear and I wish I hadn't added that feature to be honst. I may deprecate this or at least provide a better error reporting.

```python
from pycatia import catia
from pycatia.product_structure_interfaces.product_document import ProductDocument

caa = catia()
product_document: ProductDocument = caa.active_document
print(product.is_product)
```

Personally, I would just check the filename.

```python

from pycatia import catia

caa = catia()
active_document = caa.active_document
if '.CATProduct' in active_document.full_name:
    print('This is a CATProduct.')
else:
    print('This is not a CATProduct.')
```

Morpheus5828, 2025-04-30:
Ok thanks, and when it s a cat product, how did you do to going down in the tree until you will have a cat part ? I check method from Document but ...

closed 2025-06-07 by evereux