evereux/pycatia #263: [BUG] selection.select_element4 does not work as intended

CossackLucas, 2025-05-04:
**Describe the bug**
In VBA select_element4 jest returns a string to evaluate the result of selecting by the user and sets o_document to the document containing selected object. Right now it does not work that way, o_document is left as it is and the method returns tuple with string and COM object of the document which should be assigned to o_document

**To Reproduce**
Opening two CATIA documents with at least one part should show what's happening
```
def main():
        caa = catia()
        doc = catia_macro_safe_start(caa)
        doc_selection = doc.selection
        message = 'Pick a part'
        sel_filter = ('Part',)
        result_doc = Document(doc.com_object)
        response = doc_selection.select_element4(sel_filter, message, message, True, result_doc)
        response, source_doc = response
        source_doc = Document(source_doc)
        if response != 'Normal':
            return
        caa.message_box(f"Source document name is {doc.name}")
        caa.message_box(
            f"Source document name is {source_doc.name}, source item is {source_doc.selection.item2(1).value.name}.")
```

**Expected behavior**
I'm not sure, if this return is expected, I'm not that good with Python and I wasn't able to find anything in pywin32 which could explain why it returns the tuple. If it's as intended, only description should be changed to let user know, that's not only string that he has to take care of.

I wrote explicit implementation which returns str and Document object without requiring o_document (it's useless after all) and not using this strange return. I can prepare pull request if it's reproduced and fix is needed.

**Desktop:**
 - Windows 10
 - CATIA V5 R20
 - Python 3.13
 - pycatia 0.8.2

evereux, 2025-05-04:
Thanks for reporting. I'll take a look at this this week.

evereux, 2025-05-06:
Can you rewrite your reproduction example so I may run it please?

* include required imports
* remove the call to `catia_macro_safe_start`
* remove the function call
* fix indentation

CossackLucas, 2025-05-06:
Sorry, forgot to check it without my func libraries. I just checked it on IDLE, works as intended

```
from pycatia import catia
from pycatia.in_interfaces.document import Document

caa = catia()
doc = caa.active_document
doc_selection = doc.selection
message = 'Pick a part'
sel_filter = ('Part',)
response: str
result_doc = Document(doc.com_object)
response = doc_selection.select_element4(sel_filter, message, message, True, result_doc)
response, source_doc = response
source_doc = Document(source_doc)
if response == 'Normal':
    caa.message_box(f"Active document name is {doc.name}")
    caa.message_box(f"Source document name is {source_doc.name}, source item is {source_doc.selection.item2(1).value.name}.")
```

evereux, 2025-05-17:
Thanks for the updated script.

I see the issue but I'm finding it hard to understand how it should be used and thus provide a useful fix for now. The explanation in the source docs are really hard for me to grok.

I'm not giving up. I will get back to you ... 😄

CossackLucas, 2025-05-18:
If you need any additional explanation on Selection and its methods, just let me know. I had to fight with CATIA VBA long enough and I got a grasp of a thing or two ;)

evereux, 2025-05-18:
Yeah I'll definitely need to here by the looks of it. 

Let me get my head round this a little better so I can ask the question better.

evereux, 2025-08-03:
I've made the changes I think needed doing thanks to your report.

Could you take a look and let me know if this works OK for you?

evereux, 2025-08-03:
One question, what's the purpose of the first message? When I run the script I only ever see the second message.

evereux, 2025-08-14:
These changes are now available in 0.9.0.