evereux/pycatia #265: [BUG] Could not find parameter name "Sketches\Sketch_Pocket\Activity".

JiXia830, 2025-05-16:
**Describe the bug**

[2025-05-16 15:38:20,627] INFO in documents: Opening document "D:\limengting\Downloads\catia二次开发2\extra_features_of_3D_model\3D_catia_model\part1.CATPart".
Traceback (most recent call last):
  File "d:\limengting\Desktop\catia_abstract.py", line 80, in <module>
    sketch_pocket_activity = part_parameters.item("Sketches\\Sketch_Pocket\\Activity")
  File "D:\limengting\python\Lib\site-packages\pycatia\knowledge_interfaces\parameters.py", line 485, in item
    raise CATIAApplicationException(f'Could not find parameter name "{index}".')
pycatia.exception_handling.exceptions.CATIAApplicationException: Could not find parameter name "Sketches\Sketch_Pocket\Activity".

**To Reproduce**
Steps taken to reproduce the behavior:
I tried to run the example file:example__parameters__001.py
I modified the path of catia file.
e.g.
part_document: PartDocument = documents.open(Path(os.getcwd(), r"D:/limengting/Downloads/catia二次开发2/extra_features_of_3D_model/3D_catia_model/part1.CATPart"))
and then i run example__parameters__001.py

**Expected behavior**
i wanan find the example catia file  :part_measurable.CATpart,but it doesn't exist.
could you please tell me what to do to solve my problem

**Desktop (please complete the following information):**
 - OS: Windows 11
 - CATIA V5 version : CATIA(R) Version 5-6 Release 2018
 - pycatia version :latest

evereux, 2025-05-16:
`part_measurable.CATPart` is generated when running the tests but it isn't suitable for this script. I think I'd manually changed my `part_measurable.CATPart` to support this script, which I have since deleted.

The code is good from what I can tell and was originally submitted by @Luanee but there is no supporting CATPart. It was never my intention to supply supporting CATIA parts for any script.

I'll dig into this more when I get a chance and may rewrite this example or at least provide a better description in the script header.

Thanks for reporting.

JiXia830, 2025-05-21:
> `part_measurable.CATPart` is generated when running the tests but it isn't suitable for this script. I think I'd manually changed my `part_measurable.CATPart` to support this script, which I have [...]

Hi,could you please tell me how to solve the problem
"pycatia.exception_handling.exceptions.CATIAApplicationException: Could not find parameter name "Sketches\Sketch_Pocket\Activity"."

closed 2025-08-14 by evereux