evereux/pycatia #268: I hope to be able to be able to implement the new features I want

qingjiaaaa, 2025-06-16:
Hi evereux，i am a aircraft designer ,I was thrilled to find this script, and I used it and found it to work.
IN ONE OF MY WORK, I FOUND THAT I NEED TO PUT THE PARAMETERS IN THE EXCEL SHEET, AND AUTOMATICALLY CONTROL THE AUTOMATIC UPDATE OF THE CATIA FILE BY MODIFYING THE PARAMETERS OF THE EXCEL SHEET, HOW CAN I ACHIEVE THIS FUNCTION?
We look forward to hearing from you

evereux, 2025-06-16:
This is out of scope for the kind of support I have the time to provide. This is something you'll need to figure out yourself.

However, I would use something like openpyxl to read the excel sheet.

Take a look at the parameters examples provided.

If you have a specific problem regarding the usage of pycatia itself, fire away.

qingjiaaaa, 2025-06-17:
> 这超出了我有时间提供的支持类型的范围。这是您需要自己弄清楚的事情。
> 
> 但是，我会使用类似 openpyxl 的东西来阅读 excel 表格。
> 
> 查看提供的参数示例。
> 
> 如果您对 pycatia 本身的使用有具体问题，请开火。

THANK YOU VERY MUCH FOR YOUR REPLY, MAYBE I NEED TO LEARN A LITTLE MORE FROM PYHTHONN

closed 2025-06-20 by evereux