evereux/pycatia #269: [BUG] Wrong type hints for VisPropertySet methods

HubertDomaros, 2025-07-21:
**Describe the bug**
According to the type hints and docstrings, methods like `get_real_width()` or `get_real_line_type()` in `VisPropertySet` class are declared to return an int. In practice, however, these methods return a two-element tuple:
• The first element is a status code (CatVisPropertyStatus)
• The second element is the actual property value (an int)

**Steps to reproduce**
1. Open CATIA and select one or more elements (or select elements programmatically).
2. Obtain a VisPropertySet object for the selection.
3. Call get_real_width() (or get_real_line_type(), etc.).
4. Observe that the return value is a tuple (status, value) rather than a single int.
[image]

**Expected behavior**

- Type hints should return a tuple[int, int].
- The docstrings should mention the method returns (status, value), where:
    – status is a CatVisPropertyStatus enum indicating success or error
    – value is the integer property (e.g., line width or line type)

**Desktop:**
 - OS: Windows 10
 - CATIA V5 version:  V5-6R2021
 - pycatia version: pycatia 0.8.3

evereux, 2025-08-03:
I've made the changes now to the developement branch. I'll probably release in the next few days.

evereux, 2025-08-14:
0.9.0 has these changes implemented.