evereux/pycatia #275: how to use the "dimension.get_boundary_box()" in pycatia

yunzhimeng12, 2025-08-13:
we can find the "get_boundary_box() " by searching  drawing -->views-->view-->dimensions-->dimension-->get_boundary_box()   in pycatia's API .

But I use the  the  method like this,the result of print is none.
dimension11 = view.dimensions.add(D,(circle3,),(0,0,0,0),A)
dimension11.set_tolerances(1, 'ANS_NUM2', '', '', 0.1, -0.1, 0)
dimension11.move_value(pointx2+5,pointy2,0,1)
dimension11.get_dim_line().dim_line_graph_rep = 1
dimension12 = view.dimensions.add(D,(circle7,),(0,0,0,0),A)
dimension12.set_tolerances(1, 'ANS_NUM2', '', '', 0.1, -0.1, 0)
dimension12.get_dim_line().dim_line_graph_rep = 1
dimension12.move_value(pointx2+5,pointy2-16,0,1)

print(dimension11.get_boundary_box())
print(dimension12.get_boundary_box())

code is'n wrong why is the result none

yunzhimeng12, 2025-08-14:
> This is one of those class methods that need to call a VB script to execute. Normally, the class method would crash, don't think I've seen it return `None` before when this is required.
> 
> I've patched this now in the developement branch. It'll be available in 0.9.0 when released (planning on Friday).

And how to call a VB script to execute when use those class methods ,can give me a example . thank you

evereux, 2025-08-14:
You won't need to do anything differently.

yunzhimeng12, 2025-08-14:
but  now  I need to use the "dimension.get_boundary_box()" to get coordinate of  dimension's text.but it returns none

evereux, 2025-08-14:
Yes and I've patched it in 0.9.0. When it's released it'll work for you.

yunzhimeng12, 2025-08-14:
thank you very much

yunzhimeng12, 2025-08-14:
ok ,maybe you can give me effective suggestions,I want to know the coordinate of dimension's text,and than add a serial number for each dimension. Can you know other methods for adding a serial number for each dimension.

yunzhimeng12, 2025-08-14:
so   what is the different subject about,can you tell me

evereux, 2025-08-14:
This issue is related to `Dimension.get_boundary_box()`.

You have then asked about appending text to a `Dimension`. This is nothing to do with `Dimension.get_boundary_box()`.