evereux/pycatia #281: when  define_circular_detail_view is made , isn't the copy_links_to need ?

yunzhimeng12, 2025-08-16:
view2 = views.add("详图A")
behavior1 = view.generative_behavior
behavior2 = view2.generative_behavior
behavior2.define_circular_detail_view(100,55,4,behavior1 )
links1 = view.generative_links
links2 = view2.generative_links
links2.copy_links_to(links1)
view2.x = view.x+110
view2.y = view.y+35
view2.scale = 5

when up's code was executed

yunzhimeng12, 2025-08-16:
[image]

evereux, 2025-08-29:
The topic question: when define_circular_detail_view is made , isn't the copy_links_to need ?

The answer is yes, they are. If you don't do this you'll end up with an empty view. 

Are you saying that you tried this and it didn't work?

I tidied up your code sample:

* Removed unused imports
* fixed broken syntax
* renamed vars for readability
* added the line for DrawingViewGenerativeLinks.copy_links_to()

And it works fine.

```python

from pycatia import catia
from pycatia.drafting_interfaces.drawing_document import DrawingDocument

application = catia()
drawing_document: DrawingDocument = application.active_document

drawing_sheets = drawing_document.sheets
sheet1 = drawing_sheets.item('Sheet.1')

views = sheet1.views
front_view = views.item("FRONT VIEW")
front_view_genb = front_view.generative_behavior

view_detail_a = views.add("DETAIL A")
view_detail_a_genb = view_detail_a.generative_behavior
view_detail_a_genb.define_circular_detail_view(101.857925, 74.457741, 20.099856, front_view_genb)

front_view_genl = front_view.generative_links
view_detail_a_genl = view_detail_a.generative_links
front_view_genl.copy_links_to(view_detail_a_genl)

view_detail_a.x = front_view.x + 110
view_detail_a.y = front_view.y + 35
view_detail_a_genb.update()
sheet1.update()
view_detail_a.scale = 5
```

evereux, 2026-01-01:
Closing. Issue is resolved as far as I can tell.