evereux/pycatia #284: [BUG]

Morpheus5828, 2025-09-22:
```python
import os
from pycatia import catia
from pycatia.product_structure_interfaces.product import Product
from pycatia.product_structure_interfaces.product_document import ProductDocument
from pycatia.mec_mod_interfaces.part_document import PartDocument

def explore_and_export_parts(
        application,
        output_folder,
        current_product: Product
) -> None:
    """
    Recursively explores a product tree and exports each part as an STL file.
    """
    child_products = current_product.products

    if child_products.count > 0:
        for i in range(1, child_products.count + 1):
            child_item = child_products.item(i)
            child_instance = Product(child_item.com_object)
            explore_and_export_parts(
                application=application,
                output_folder=output_folder,
                current_product=child_instance,
            )
    else:
        instance_name = current_product.name
        safe_filename = "".join(c for c in instance_name if c.isalnum() or c in (' ', '.', '_')).rstrip()
        output_file = os.path.join(output_folder, f"{safe_filename}.stl")

        print(f"Processing Part: {instance_name} ...")

        try:
            ref_product = current_product.reference_product
            part_document = PartDocument(ref_product.parent.com_object)

            part_document.new_window()

            part_document.export_data(output_file, "stl")

            part_document.close()

            print(f"--> Successfully exported to {output_file}")

        except Exception as e:
            print(f"--> FAILED to export {instance_name}. Error: {e}")

if __name__ == "__main__":
    caa = catia()

    active_doc = caa.active_document
    output_file_path = "C://Users//.../Desktop//STL_OUT//"

    product_document = ProductDocument(active_doc.com_object)
    root_product = Product(product_document.product.com_object)

    explore_and_export_parts(
        output_folder=output_file_path,
        application=caa,
        current_product=root_product,
    )

```

Morning, 
I ve got some trouble to save a catpart as stl file.
This code give me no bhug, but when I check my destination folder, I ve got no STL files.
In my CATIA, I ve got a tree of catproduct, it the reason why, I m using recursivity.
In my company, we had created a Dassault Ticket to explore why we could't extract stl directly by the tree.
They say, we have to open each catpart in a new catproduct.
But i m still in the cloud ...
Best regards

evereux, 2025-09-22:
I have changed the Issue title to something more descriptive than just BUG.

You've posted your recursive version. Have you tried with just a single part? Does that work? Do you have the license to be able to export to STL?

evereux, 2025-10-22:
> When creating the new ticket please also let me know which workbench the tool is in.

Figured it out. It's a sub-menu of the DMU Sectioning tool I've never used.